prusa3d/PrusaSlicer #3748: BugFix : filling 100% crash prusaslicer

Mathes67, 2020-03-01:
### Version
PrusaSlicer 2.2.0-rc
MAC : PrusaSlicer-2.2.0-rc+-202002290955

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
MAC OSX 10.15.3 

### Behavior
Prusa slicer crash when infill 100 % is select

Open prusa slicer
Change infill 100%
= Crash

(Change infill to rectiligne and after change % of infill, prusa slicer don't crash)

[image]

lukasmatena, 2020-03-02:
This is translation-related bug (one percent sign instead of two). It affects Spanish, French and Italian translations. We should fix this.

lukasmatena, 2020-03-02:
Fixed by b95d635.
Thanks for the report.

Closing.

closed 2020-03-02 by lukasmatena