prusa3d/PrusaSlicer #3749: searching for slicer help - 404

3Dietrich, 2020-03-01:
### Version
2.1.1+
### Operating system type + version
MacOS Sierra 10.12.6
### 3D printer brand / version + firmware version (if known)
MK3S
### Behavior
#### _Describe the problem_
Often I am desperate to find some Slicer help. I know you just did it very well but.. there is no Link anyware. Today I thought 'what about a help Link in Slicer?' Ha! There is Slicer/Help/PrusaSlicer Website! But you know what this leads me to?  **404** :(
#### _Steps needed to reproduce the problem_
open Slcer, call: Help/PrusaSlicer Website
#### _Expected Results_
finally a clear way to find Prusa Slicer Help/Manual
#### _Actual Results_
Uh Oh 404
#### _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]
[image]

3Dietrich, 2020-03-01:
Well, I just found it by searching the url a bit.
it's https://www.prusa3d.com/prusaslicer/

hopefully I remember this (actually easy) weblink next time..

bubnikv, 2020-03-02:
Thanks for heads up. I will notify the content team.

Blixak, 2020-03-02:
Redirect is fixed now. And btw, you can find the brand new PrusaSlicer documentation here: https://help.prusa3d.com/en/article/general-info_1910 (we will add the link to the product page too). Thank you!

bubnikv, 2020-03-03:
@Blixak thanks.
Closing.

closed 2020-03-03 by bubnikv