prusa3d/PrusaSlicer #3750: Holes are not being sliced

ossimc82, 2020-03-01:
### Version
Version 2.2.0-rc+win64

### Operating system type + version
Windows 10 Pro Version 1903

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S (Purchased Feb. 2020) FM: 3.8.1-2869

### Behavior
* _Holes are not being sliced_
* _No holes in the gcode preview_

[image]

* _Just a flat plane completly filled_

[image]

* If you export the project to .3mf and reopen it in prusa slicer, it will slice correctly

### Steps to reproduce
* Open Prusa Slicer
* Import provided .stl from the zip
* Change settings:
> Printer: Original Prusa i3 MK3S
> Filament: Prusament PLA
> Print Settings: 0.15mm Quality
* Click slice

#### Stl File
* [top.zip](https://github.com/prusa3d/PrusaSlicer/files/4272527/top.zip)

rtyr, 2020-03-02:
Repair the model by "Fix through the Netfabb" option. It should fix this issue.

ossimc82, 2020-03-02:
Yeah it does fix it, weird tho that it doesnt show an exclamation mark next to the model in the object list.

bubnikv, 2020-03-02:
We are marking these and similar issues with "model import" flag, so we can
look into all these models that refuse to slice correctly in a single shot.

closed 2026-09-29 by prusaslicer-bot[bot]