prusa3d/PrusaSlicer #3751: Inconsistent "Open" vs. "Import" behavior

kssi, 2020-03-02:
### Version
Version: 2.2.0-rc+
Build: PrusaSlicer-2.2.0-rc+-202002290955

### Operating system type + version
Operating System: Macintosh
System Architecture: 64 bit
System Version: OS X El Capitan Version 10.11.6 (Build 15G22010)

### Behavior
* _Describe the problem_

On macOS, when double-clicking a .3mf file in the Finder, PrusaSlicer imports its content instead of _opening_ the actual project.

If PrusaSlicer **is not** currently running, the content of the .3mf file is imported into a new blank project. As a result, "File > Save Project" doesn't save the modifications into the (supposedly) opened project but prompts the user with a "Save Project as..." dialog.

If PrusaSlicer **is** currently running and a project is already opened, the content of the .3mf file is imported into the current project, thus merging two completely unrelated files together.

* _Expected Results_

Opening from the Finder should behave exactly like "File > Open Project...".

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

No screenshot needed, but this can be seen in the title bar of the application. The name of the project shows up only when actually opened.

kssi, 2020-03-02:
#2722

kssi, 2020-03-02:
This might be related: #2722

brettvitaz, 2020-03-02:
This behavior was counter-intuitive at first, but now is a part of my workflow as it makes adding models to the built plate easier. Otherwise any time you open a 3mf file from finder it would wipe out your current settings, and having different behavior when the application is closed would be even more confusing. The mental hurdle that needs to be overcome is ensuring that you do project level operations from the menu, starting with an Open or New, but I think it was a good design decision.

kssi, 2020-03-05:
> This behavior was counter-intuitive at first, but now is a part of my workflow as it makes adding models to the built plate easier.

Considering PrusaSlicer as analogous to a physical built plate is a broken approach. Softwares are not meant to mimic the physical world, that's not how they work. As a side note, PrusaSlicer not being able to handle multiple documents at the same time is highly questionable.

> Otherwise any time you open a 3mf file from finder it would wipe out your current settings

.3mf was designed to store entire, self-contained projects. They are documents. They store ALL settings. So wiping out the current settings is exactly what you want when you open them.

> but I think it was a good design decision.

As long as .3mf and other self-contained project files are concerned, that's a bug, not a design decision. You are confusing them with asset files such as STL, which are definitely meant to be _imported_.

Finally, when I right-click a file in the Finder I see "Open", not "Import".

Would you like Microsoft Word to merge your Curriculum Vitæ with your shopping list when you double-click shopping-list.word?

brettvitaz, 2020-03-06:
> Considering PrusaSlicer as analogous to a physical built plate is a broken approach.

As professional software engineers, we build software that considers the user's and their workflow, and sometimes mimicking the real world is the exact right thing.

> As long as .3mf and other self-contained project files are concerned, that's a bug, not a design decision.

3mf can contain many things, and one of them is a self contained project. Your example is not a good one. This is certainly not a bug, it is a design choice, and one that meets the use case very well.

> Would you like Microsoft Word to merge your Curriculum Vitæ with your shopping list when you double-click shopping-list.word?

Word opens a completely new window to accommodate. Not every tool in your tool box needs to behave identically, and the use case must be considered. We are filling a build plate, not composing a document.

Sembazuru, 2020-05-03:
I would argue that the behavior of double clicking on a file in the file system treated as an import instead of an open IS counter-intuitive because it breaks existing convention. The example above referencing Word is exactly how I would expect PrusaSlicer or any other program to function. I.e. the new file gets opened in what ever the application's mode if a new instance is handled (new window, new tab, etc) not to have the new file merged with an already open file. If the user wants to import one file into another, there are already existing workflows, namely a File/Import menu item and copy-paste from one program instance to another.

This has been true since at least my first experience with windowing GUIs (Commodore Amiga back in 1985). Any other behavior is now non-intuitive.

lukasmatena, 2024-02-18:
When double clicking a 3mf file, PrusaSlicer now (as of 2.7.2-alpha2, but it was implemented much sooner) shows a dialog asking the user about how it should behave (open project / import geometry). Closing.

closed 2024-02-18 by lukasmatena