prusa3d/PrusaSlicer #3753: Prusament PETG filament option is removed from dropdown after changing nozzle size

yellowbeak, 2020-03-02:
### Version
Version 2.2.0-rc+linux-x64

### Operating system type + version
1 SMP Debian 4.19.67-2+deb10u2 (2019-11-11)

### 3D printer brand / version + firmware version (if known)
Prusa Mk3S
prusa3d_fw_MK3S_3_8_1_2869.hex

### Behavior
Using the standard machine settings of 'Original Prusa i3 MK3S', I can select the filament type 'Prusament PETG'.
If I change the 'Nozzle diameter' in extruder from 0.4 to 0.6, then change the 'Print settings' to anything different, for instance '0.3mm DRAFT', then the 'Filament' automatically changes from the previously selected 'Prusament PETG' to 'Prusament PLA'.
If I click on the 'Filament' dropdown, then there is no option for 'Prusament PETG' - it is absent.
If I click 'Add/Remove Filaments', then select 'PETG' -> 'Prusa Poylymers', then 'Prusament PETG' is shown as an option, but is already ticked.  If I click 'Finish', then this filament type is still not shown in the filament dropdown menu, so cannot be selected or used.

ochm, 2020-03-02:
Because in the filament profile is set condition for nozzle diameter? It is not a bug! It is feature.

yellowbeak, 2020-03-03:
> Because in the filament profile is set condition for nozzle diameter? It is not a bug! It is feature.

Thanks ochm.  I've found where that's now.

closed 2020-03-03 by yellowbeak