prusa3d/PrusaSlicer #3754: Banding round perimeter of prints when using 'no sparse layers' option

alienryes, 2020-03-02:
### Version
PrusaSlicer-2.2.0-beta+win64

### Operating system type + version
Windows 10 Pro Insider Preview 2004 build 19564.1000

### 3D printer brand / version + firmware version (if known)
Prusa MK3S with MMU2S

### Behavior
Intermittently, when a print is made with the 'No sparse layers' option enabled the print displays bands of slightly thicker perimeter all the way round at certain heights. This has happened when using both Octoprint via USB and when printing direct from SD card.

[image]

[image]
[image]

To me it looks like when the extruder returns to the print from the wipe tower it is slightly lower than it should be and so the layer gets 'squashed' more than the other layers and is pushed outward slightly. It does not appear to be layer shifting as it encircles the whole print and is not just in one direction.

#### Project File (.3MF) where problem occurs
[omnom_body.zip](https://github.com/prusa3d/PrusaSlicer/files/4277178/omnom_body.zip)

bubnikv, 2020-03-17:
It may be, that your nuts on your Z screws are not seated correctly or the Z axes are not straight, so they do not move smoothly?

Maybe we need to lower the deceleration on the Z axis?

Most likely it is a mechanical issue, but we shall retest in house.

bubnikv, 2020-03-17:
Looks like this is a duplicate of #3842. I am closing this issue, let's continue at #3842.

closed 2020-03-17 by bubnikv