prusa3d/PrusaSlicer #3756: Volume purge are not saved

Forfelet, 2020-03-02:
### Version
Version 2.2.0-rc+linux-x64

### Operating system type + version
Windows 7 64bits

### 3D printer brand / version + firmware version (if known)
Prusa MK3S + MMU2S

### Behavior
When I change the type of printer, my purge volumes are reset.

I have one MK3S and one MK3S + MMU2S.
When change printer profile from MK3S to MK3S + MMU2s and vice versa, values are reset (unloading 70, loading 70 for 5 filaments)
So, can you saved these values in the printer profile or registry Windows ? 
Because, when I restart PrusaSlicer, I must set these values yet. It's so painful to change these values all the time.

Many thanks for your job.
I hope that you will include this proposal in the new version.

Best regards,
Cyril

VanessaE, 2020-03-15:
This issue and #3581 #2990 #1282 #1693 all seem to be duplicates of one another, or anyway dupes of the underlying idea, that purge volumes aren't saved in a user-friendly way.

In my opinion, purge volumes need to be saved and loaded in such a way that the user can build up a full table of relationships over time, to eventually cover all of the filaments he or she uses.  Personally, I favor storing the table in the printer profile; the user can then just adjust the relationships every so often i.e. when a new filament needs to be added to the table, and then just re-save the printer profile.

Storing it there makes the most sense, since different hardware (I.e. one hotend vs. another, or direct-feed vs. bowden-fed) will affect purging behavior.

lukasmatena, 2024-03-21:
[PrusaSlicer 2.7.3-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-alpha1) introduced a possibility to set the purging volumes in Printer / Filament Settings. Closing.

closed 2024-03-21 by lukasmatena