prusa3d/PrusaSlicer #3758: Ender 3 Profile - Remaning Time

RivasBoy, 2020-03-02:
### Version
PrusaSlicer 2.2.0-rc

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
windows 10 pro

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro / board fysetc cheetah 1.2 MARLIN BUG FIX 2.0X

### Behavior
The remaning time of the priting job  does not work.

bubnikv, 2020-03-03:
>  The remaning time of the priting job does not work.

What do you mean?

neophyl, 2020-03-03:
I don't have an Ender 3 but I do have a CR10 running Marlin and if the Ender is like the cr10 then the firmware doesn't know how to deal with the time remaining gcode so it 'doesn't work' in the sense that the time remaining is not displayed on the printers lcd.

However as its the firmware and not the slicer it wasn't anything to post about here.  This may *not* be the same with the Ender but I thought I should mention the possibility.

RivasBoy, 2020-03-03:
Good Morning,
in the ender 3 pro with the slicer cura the post processing script show remaning time in lcd, works for me.

In prusa slicer in the tab Printer Settings I check the option Supports remaining times:
I think this is the right way to do it.

neophyl, 2020-03-03:
Can you compare the gcode for the 2 slicers ?  That may help to figure out what the difference is.

RivasBoy, 2020-03-03:
Ok. 
when i arrived at home i will compare the booth files.
Thank you.

RivasBoy, 2020-03-03:
Hello,
Cura slicer added to file the code "M117 Time Left 0h33m48s", several times.
For me the cura slicer calculate the time by layer and add this command during the print job.

neophyl, 2020-03-03:
Ah that possibly explains it.  M117 is the general print xxxxx to screen command.
I'm sure Bubnik will correct me if I'm wrong :) but I think prusa slicer uses the M73 - set print progress command, for example :-
M73 Q2 S41 - so time remaining in stealth mode 41 (S41)
M73 P2 R41 - time remaining in normal mode 41 (R41)
as I think the Prusa firmware is programmed to respond to those and can give time remaining in either normal or stealth modes.

At least when I've looked through any of my gcode files those commands seem to correspond to the time remaining displayed on the lcd at that point.

bubnikv, 2020-03-04:
yes, the remaining times is a Prusa firmware specific feature. It emits a
G-code that only Prusa firmware understands.

neophyl, 2020-03-04:
I don't know anything about writing post processing scripts but I know Prusa Slicer can run them, so it should be possible to script something to replace/addto the M73 gcode lines with M117 Time Left and use the Rxx part to supply the time part ?

bubnikv, 2020-03-04:
that will certainly work.

RivasBoy, 2020-03-04:
I will study post-processing scripts and contribute if I can.

bubnikv, 2020-03-04:
This may help if you are on Windows and you are brave to try Perl. On OSX
or Linux the Perl interpreter is installed by default.
https://github.com/prusa3d/PrusaSlicer/wiki/Running-Perl-post-processing-scripts-on-Windows-from-Slic3r-PE-1.42-and-newer

This Perl install has been derived from our old Perl based Slic3r Prusa
Edition and it is tiny, it consumes just 6.64 MB of your hard drive.

This may help as well.
https://manual.slic3r.org/advanced/post-processing
https://hackaday.com/2016/07/20/3d-printering-g-code-post-processing-with-perl/

rtyr, 2021-02-08:
I believe this issue can be closed. Also, as far as I know, the current Marlin supports our remaining time format so you may just update the firmware.

closed 2021-02-08 by rtyr