prusa3d/PrusaSlicer #3759: Error loading 3MFs in 2.2.0 RC

3Delights, 2020-03-03:
### Version
Version 2.2.0-rc+win64

### Operating system type + version
Windows 10 Pro x64 Fully updated
### PrusaSlicer
Version:   2.2.0-rc+win64
Build:     PrusaSlicer-2.2.0-rc+win64-202002291006

### Sys Info
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 68,634MB
OpenGL installation
GL version:   4.6.13587 Compatibility Profile Context 20.2.1 26.20.15019.1003
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon RX 5700 XT
GLSL version: 4.60
Installed extensions:
GL_AMDX_debug_output
GL_AMD_blend_minmax_factor
GL_AMD_conservative_depth
GL_AMD_debug_output
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_sample_positions
GL_AMD_gcn_shader
GL_AMD_gpu_shader_half_float
GL_AMD_gpu_shader_half_float_fetch
GL_AMD_gpu_shader_int16
GL_AMD_gpu_shader_int64
GL_AMD_interleaved_elements
GL_AMD_multi_draw_indirect
GL_AMD_name_gen_delete
GL_AMD_occlusion_query_event
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_sample_positions
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_atomic_counter_ops
GL_AMD_shader_stencil_export
GL_AMD_shader_stencil_value_export
GL_AMD_shader_trace
GL_AMD_shader_trinary_minmax
GL_AMD_sparse_texture
GL_AMD_sparse_texture_pool
GL_AMD_stencil_operation_extended
GL_AMD_texture_cube_map_array
GL_AMD_texture_texture4
GL_AMD_transform_feedback3_lines_triangles
GL_AMD_transform_feedback4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
[... 247 lines cut ...]
GL_OES_EGL_image
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK2 Official Kit (No mods) Firmware = Latest

### Behavior
The attached 3MF file loads fine in the last stable version of PrusaSlicer (PrusaSlicer-2.1.1-win64-20912101512 but causes the latest (RC) to crash with a list of error messages:

[image]

There are 21 errors listed in all.

Clicking Exit closes the program, clicking Continue causes the error dialog to reappear, this repeats 3 or 4 times then the object loads but PrusaSlicer is corrupted (Eg. the 3D bed image disappears).  This was happening with SEVERAL 3MFs on the Beta versions but the RC version seems to only do it with a few files now

#### Project File (.3MF) where problem occurs
[Column 8 v6 test.zip](https://github.com/prusa3d/PrusaSlicer/files/4278973/Column.8.v6.test.zip)

#### Profile folder
[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/4278977/PrusaSlicer.zip)

bubnikv, 2020-03-03:
I cannot reproduce it straight away on my computer even using your config.
I have a feeling it may be dependent on the screen resolution. What is the
display scaling of the screen the PrusaSlicer runs on?

bubnikv, 2020-03-03:
Luckily @YuSanka is able to reproduce the issue, so we can fix it. Thanks for heads up.

YuSanka, 2020-03-03:
Please, check this [build ](https://drive.google.com/file/d/1wBgowfHQK23yaUF7NgZBssEy53RBnoMl/view?usp=sharing)

3Delights, 2020-03-03:
That new build now appears to open all my files again, but on some of them (including the one I posted originally) the 3D image of the print bed is missing:

[image]

Here is another problem 3MF: 
[Column 7b2.zip](https://github.com/prusa3d/PrusaSlicer/files/4281875/Column.7b2.zip)

Where as this one works fine: 
[Column 7b.zip](https://github.com/prusa3d/PrusaSlicer/files/4281890/Column.7b.zip)

bubnikv, 2020-03-04:
> That new build now appears to open all my files again, but on some of
them (including the one I posted originally) the 3D image of the print bed
is missing:
Because the printer profile stored in this 3mf does not refer to a Prusa3D
system profile.

[image: image.png]

3Delights, 2020-03-04:
> Because the printer profile stored in this 3mf does not refer to a Prusa3D system profile.

Hmmm, they were both saved in the same way with roughly the same settings.  They load OK in the old stable release as well.  I only have the one printer profile because that's all I've got - a Prusa i3 MK2.

Is there a way to correct this?  I can open up the 3MF and edit the Metadata files.

bubnikv, 2020-03-05:
I don't know what happened. You may rename the 3mf to zip, unzip, edit Column 7b2\Metadata\Slic3r_PE.config
Here you will find a key
; inherits_cummulative = "0.20mm NORMAL";"Prusament PLA";
just add the name of your printer profile at the end, zip it back and rename to 3mf.

bubnikv, 2020-03-07:
The original issue will be fixed in PrusaSlicer 2.2.0-rc2. Closing.

closed 2020-03-07 by bubnikv