prusa3d/PrusaSlicer #3761: [2.2.0-rc+] Configuration Assistant doesnt work as expected with "Other Vendors"

tilllt, 2020-03-03:
<details>
 <summary>System Info (click to expand)</summary>
PrusaSlicer
Version:   2.2.0-rc+
Build:     PrusaSlicer-2.2.0-rc+-202002290955

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Sierra Version 10.12.6 (Build 16G2016)
Total RAM size [MB]: 8,590MB
OpenGL installation
GL version:   2.1 NVIDIA-10.18.5 378.05.05.25f16
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce 9500 GT OpenGL Engine
GLSL version: 1.20
Installed extensions:

GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_depth_bounds_test
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_fragment_program2
GL_NV_fragment_program_option
GL_NV_light_max_exponent
GL_NV_multisample_filter_hint
GL_NV_point_sprite
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod

</details>

Not sure if this is intended behaviour. When starting from scratch by selecting "Configuration Assistant" -> Remove User Profiles, adding "Other Vendors", "Creality FFF", Adding All Filaments, then starting the Program, Prusaslicer starts with "Original Prusa i3 MK3" selected. No big deal, i can switch to the Ender3 manually, but my expecation would be it starts with the Printer i selected in the "Configuration Assistant", not the i3 MK3.

Also, the Filaments i selected are not added to the configuration.
 
See Video for illustration:
https://pixelscope.de/videos/watch/50ef12d6-4ba4-4d45-81ce-9340627db9b1

bubnikv, 2020-03-04:
> System Info (click to expand) PrusaSlicer Version: 2.2.0-rc+ Build:

Frankly we had to dig deep into our memories to try to remember what was
the intended behavior. The checkbox currently seemingly just removes user's
profiles. Maybe we will just update the checkbox label.

The other issue is that you jumped over the Prusa specific printer
selection pages in the wizard and some of the printers were installed
without you knowing it. We did not find any good solution to that problem.
Do you have any proposal?

tilllt, 2020-03-04:
Well, since Prusa pays for the development if this Fork (thanks btw.) i can understand that you preselect the most sold Prusa Printer. In theory it would make more sense to not preselect anything imho OR as the final page of the Wizard make a Summary page which lists all the things you are going to get when you press OK. Each item with a little Trashcan to remove it - OR a link to go back to the respective config page. 

Still, shouldnt all Filaments be added if i go through the Wizard like i did (see Video)?

neophyl, 2020-03-04:
If you only select a custom printer then the filaments list isn't available at all (its skipped) in the configuration wizard.  
The filaments are part of the Vendor space and so not available to custom printers.

tilllt, 2020-03-04:
Well, from a UX point of view that makes no sense and is not understandable at all. If the Filament presets are tied to installing a certain Printer, it should say so on the Filament selection page AND there have to be Filament selection Tabs for each printer, for example.

neophyl, 2020-03-04:
For further information please see the discussion on the Prusa Forum here https://forum.prusaprinters.org/forum/prusaslicer/prusaslicer-2-2-0-beta-2/
and issue 3658

tilllt, 2020-03-04:
Well, what i can gather from that thread is that other people are struggling to understand the behaviour exactly the same way that i do ;)

bubnikv, 2020-03-15:
The checkbox title was confusing. I have removed the "install from scratch" note. The checkbox really just removes the user profiles, it does not make any change in installation of system profiles.
The change will be in PrusaSlicer 2.2.0-rc4. Closing.

closed 2020-03-15 by bubnikv