prusa3d/PrusaSlicer #3762: Bridging not detected for infill

Anima1000000, 2020-03-03:
### Version
PrusaSlicer-2.2.0-rc+win64

### Operating system type + version
Windows 10 1909 OS build 18363.657

### CR10S4 Marlin 1.1.9 MKS Gen L 1.0

### Behavior
Bridges are detected for perimeters but not for infill.

[image]

#### Project File (.3MF) where problem occurs
[EndstopZ.zip](https://github.com/prusa3d/PrusaSlicer/files/4281594/EndstopZ.zip)

bubnikv, 2020-03-04:
You have no bridges detected because you set the number of bottom layers to
zero.

Anima1000000, 2020-03-05:
Thx, Solid part didn't think it would need a bottom layer

closed 2020-03-05 by Anima1000000