prusa3d/PrusaSlicer #3764: 2.2.0-rc can't slice a model in vase mode

si14, 2020-03-03:
### Version
2.2.0-rc

### Operating system type + version
Mac OS Catalina

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior

The output of the slicer is clearly wrong in vase mode. See the screenshot:

[image]

The model is `pyramid5.stl` from here: https://www.thingiverse.com/thing:1356547/files

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs

[pyramid5.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4281734/pyramid5.3mf.zip)

tilllt, 2020-03-04:
Just out of curiosity, can other slicers correctly slice this in vase mode?

si14, 2020-03-04:
Yes, 2.1.0 does slice it more or less correctly
[image]

There are a few weird "missing" layers (actually there is material there in narrow bands) 
[image] but it's definitely more correct than 2.2.0-rc

bubnikv, 2020-03-04:
I believe PrusaSlic3r 2.2.0 is fine. 

The way every slicer calculates perimeters is it slices the model, then it shrinks the resulting polygon by half the perimeter width. This shrinking operation will split the polygon into multiple pieces, which itself contradicts the spiral vase mode.

You just scaled the model too small so that the model is no more printable. The only difference between PrusaSlicer 2.2.0 and 2.1.1 here is that PrusaSlicer 2.2.0 removes all but the largest contour at each layer while PrusaSlicer 2.2.1 keeps them and then the physical print fails.

bubnikv, 2020-03-15:
@si14 Is the explanation clear? Could you close the issue?

rtyr, 2020-04-06:
No response. Closing.

closed 2020-04-06 by rtyr