prusa3d/PrusaSlicer #3766: 2.2.0-rc Goes into hanging loop upon opening

kenshinvt, 2020-03-03:
### Version
2.2.0-rc 64 bit release windows

### Operating system type + version
Win10 Pro latest updates
GTX 1080 video card latest nvidia drivers (clean reinstall tested).

### Behavior
During normal use (adding supports, manipulating model in SLA gizmo) prusaslicer will hang with loading icon at a random point.  Force close necessary.  

After this happens, restarting Prusaslicer will cause the same issue.  The app will start and hang with a loading icon.  Picture of it frozen below. Sometimes the loading icon will disappear for a fraction of a second then will return.  Frozen/cannot interact with menus and must force close.  Any starts to Prusaslicer.exe in that windows session will produce the same result.

Restarting Windows will resolve the issue temporarily.  Following things tested before posting here:

--Installing 2.1.1 installer and running does not show the same issue.  Runs normally.  Even after the hang has occurred in that windows session.
--v2.2 beta/alpha (and also 32 bit versions) show the same behavior.  However, I used the alpha and beta releases of 2.2.0 and did not start seeing this issue until starting to use 2.2.0 RC.  I have been using prusaslicer past releases for ~1.5yrs and have never seen this issue in any 2.1x or below versions.
--It is not related to system audio drivers as seen in some other feedback and posts.  I don't have those specific audio drivers but have also tested by stripping down to minimal drivers and even disabled sound and display devices.  I have also tried stopping all 3rd party apps down to windows core services to rule out 3rd party interference.
--Turning on software rendering does not impact/stop it.  I tried both the console command line SW-render option and also the opengl32.dll moved to folder with executable.  If I do these, I see a lower res (aliasing visible) print bed on startup and then the freezing/hanging starts.
--deleting the prusaslicer userdata folder (so it is recreated on startup) does not have any impact.  New install's prusaslicer userdata folder does not help either.

[image]

#### Project File (.3MF) where problem occurs
Specific file not involved.  This happens when starting the program with no input.

bubnikv, 2020-03-04:
Your issue may be related to #3515 #3733 #3746
We have a hypothesis what causes the issue and we are implementing a workaround as of now. We will send out a binary build for you guys to test once we are ready.

bubnikv, 2020-03-07:
We have reworked two services that were introduced in PrusaSlicer 2.2.0-alpha: Detection of 3DConnexion USB plug / unplug events and removable USB driver or SD card plug / unplug events. These detection tasks were performed at a user interface thread, while now they are executed in their respective background threads in PrusaSlicer 2.2.0-rc2. Plase let us know whether this refactoring resolved your issues. We are closing this issue. Please feel free to reopen if your issue has not been resolved with PrusaSlicer 2.2.0-rc2.

closed 2020-03-07 by bubnikv