prusa3d/PrusaSlicer #3767: wxwidgets breaking linux build

clytle374, 2020-03-04:
_version_2.2.0-rc-17-ga87ba5d6a  Seems that commit 8b16b2c12e241919984558134e7af7f571c26d65
is causing Gentoo Linux build failure with the following messages.  Let me know if I can provide more information or test something. I am using wxwidgets 3.0 as 3.1 isn't on Gentoo yet.  
Thanks
Cory
`[ 77%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/SysInfoDialog.cpp.o
/home/cory/sources/prusa/src/slic3r/GUI/AboutDialog.cpp: In constructor ‘Slic3r::GUI::CopyrightsDialog::CopyrightsDialog()’:
/home/cory/sources/prusa/src/slic3r/GUI/AboutDialog.cpp:41:93: error: cannot convert ‘boost::basic_format<char>::string_type’ {aka ‘std::__cxx11::basic_string<char>’} to ‘const char*’
   39 |     : DPIDialog(NULL, wxID_ANY, wxString::FromUTF8((boost::format("%1% - %2%")
      |                                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~               
   40 |                                                        % SLIC3R_APP_NAME
      |                                                        ~~~~~~~~~~~~~~~~~                     
   41 |                                                        % _utf8(L("Portions copyright"))).str()),
      |                                                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
      |                                                                                             |
      |                                                                                             boost::basic_format<char>::string_type {aka std::__cxx11::basic_string<char>}
In file included from /usr/include/wx-3.0/wx/memory.h:15,
                 from /usr/include/wx-3.0/wx/object.h:19,
                 from /usr/include/wx-3.0/wx/event.h:16,
                 from /usr/include/wx-3.0/wx/app.h:19,
                 from /home/cory/prusaslic3r/src/slic3r/libslic3r_gui_pch/pchheader.hpp:95:
/usr/include/wx-3.0/wx/string.h:1742:42: note:   initializing argument 1 of ‘static wxString wxString::FromUTF8(const char*, size_t)’
 1742 |     static wxString FromUTF8(const char *utf8, size_t len = npos)
      |                              ~~~~~~~~~~~~^~~~
/home/cory/sources/prusa/src/slic3r/GUI/AboutDialog.cpp: In constructor ‘Slic3r::GUI::AboutDialog::AboutDialog()’:
/home/cory/sources/prusa/src/slic3r/GUI/AboutDialog.cpp:201:111: error: cannot convert ‘boost::basic_format<char>::string_type’ {aka ‘std::__cxx11::basic_string<char>’} to ‘const char*’
  201 |     : DPIDialog(NULL, wxID_ANY, wxString::FromUTF8((boost::format(_utf8(L("About %s"))) % SLIC3R_APP_NAME).str()), wxDefaultPosition,
      |                                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
      |                                                                                                               |
      |                                                                                                               boost::basic_format<char>::string_type {aka std::__cxx11::basic_string<char>}
In file included from /usr/include/wx-3.0/wx/memory.h:15,
                 from /usr/include/wx-3.0/wx/object.h:19,
                 from /usr/include/wx-3.0/wx/event.h:16,
                 from /usr/include/wx-3.0/wx/app.h:19,
                 from /home/cory/prusaslic3r/src/slic3r/libslic3r_gui_pch/pchheader.hpp:95:
/usr/include/wx-3.0/wx/string.h:1742:42: note:   initializing argument 1 of ‘static wxString wxString::FromUTF8(const char*, size_t)’
 1742 |     static wxString FromUTF8(const char *utf8, size_t len = npos)
      |                              ~~~~~~~~~~~~^~~~
/home/cory/sources/prusa/src/slic3r/GUI/AboutDialog.cpp:284:36: error: cannot convert ‘boost::basic_format<char>::string_type’ {aka ‘std::__cxx11::basic_string<char>’} to ‘const char*’
  265 |             (boost::format(
      |             ~~~~~~~~~~~~~~~         
  266 |             "<html>"
      |             ~~~~~~~~                
  267 |             "<body bgcolor= %1% link= %2%>"
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  268 |             "<font color=%3%>"
      |             ~~~~~~~~~~~~~~~~~~      
  269 |             "%4% &copy; 2016-2019 Prusa Research. <br />"
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  270 |             "%5% &copy; 2011-2018 Alessandro Ranellucci. <br />"
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  271 |             "<a href=\"http://slic3r.org/\">Slic3r</a> %6% "
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  272 |             "<a href=\"http://www.gnu.org/licenses/agpl-3.0.html\">%7%</a>."
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  273 |             "<br /><br />"
      |             ~~~~~~~~~~~~~~          
  274 |             "%8%"
      |             ~~~~~                   
  275 |             "<br /><br />"
      |             ~~~~~~~~~~~~~~          
  276 |             "%9%"
      |             ~~~~~                   
  277 |             "</font>"
      |             ~~~~~~~~~               
  278 |             "</body>"
      |             ~~~~~~~~~               
  279 |             "</html>") % bgr_clr_str % text_clr_str % text_clr_str
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  280 |             % copyright_str % copyright_str
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  281 |             % is_lecensed_str
      |             ~~~~~~~~~~~~~~~~~       
  282 |             % license_str
      |             ~~~~~~~~~~~~~           
  283 |             % based_on_str
      |             ~~~~~~~~~~~~~~          
  284 |             % contributors_str).str());
      |             ~~~~~~~~~~~~~~~~~~~~~~~^~
      |                                    |
      |                                    boost::basic_format<char>::string_type {aka std::__cxx11::basic_string<char>}
In file included from /usr/include/wx-3.0/wx/memory.h:15,
                 from /usr/include/wx-3.0/wx/object.h:19,
                 from /usr/include/wx-3.0/wx/event.h:16,
                 from /usr/include/wx-3.0/wx/app.h:19,
                 from /home/cory/prusaslic3r/src/slic3r/libslic3r_gui_pch/pchheader.hpp:95:
/usr/include/wx-3.0/wx/string.h:1742:42: note:   initializing argument 1 of ‘static wxString wxString::FromUTF8(const char*, size_t)’
 1742 |     static wxString FromUTF8(const char *utf8, size_t len = npos)
      |                              ~~~~~~~~~~~~^~~~
[ 77%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/KBShortcutsDialog.cpp.o
make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/build.make:89: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/AboutDialog.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
`

lukasmatena, 2020-03-04:
Thanks. Maybe the same problem as in #3745, just reported better this time. We will fix this.

clytle374, 2020-03-04:
Sorry to double up bug reports, I had no idea what the other build failure was.  Let me know if I can test anything for you all.  My work schedule is non-normal, but will do my best.
Cory

lukasmatena, 2020-03-04:
> Sorry to double up bug reports

That's not how I meant it, you could hardly have known what the other issue was. I added the reference to link the issues together, not to tell you off.

>  Let me know if I can test anything for you all.

Thank you very much. I managed to reproduce the problem, so it probably won't be necessary. But I will let you know.

clytle374, 2020-03-04:
> That's not how I meant it, you could hardly have known what the other issue was. I added the reference to link the issues together, not to tell you off.

I did not take it that way.  I just hate being the person that doesn't search before complaining.

lukasmatena, 2020-03-04:
It should be fixed on master by https://github.com/prusa3d/PrusaSlicer/commit/eb3b65d8afcc7026cec0b2054152d78e08673041.

clytle374, 2020-03-04:
Builds, runs, slices, closed
Thanks
Cory

closed 2020-03-04 by clytle374