prusa3d/PrusaSlicer #3768: Protocol "https" not supported or disabled in libcurl

ondratu, 2020-03-04:
### Version
PrusaSlicer-2.2.0-rc+linux-x64-202002291012

### Operating system type + version
Operating System: Unix
System Architecture: 64 bit
System Version: Linux 5.4.0-4-amd64 x86_64
Total RAM size [MB]: 16,549MB
(Debian Linux)

### Behavior
Just start the appimage. Here is part of log
```
[2020-03-04 10:33:20.289588] [0x00007fa778dfc700] [error]   Error getting: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`: HTTP 0, Unsupported protocol:
Protocol "https" not supported or disabled in libcurl
[Error 1]
[2020-03-04 10:33:20.291028] [0x00007fa77ac57540] [error]   Unable to enumerate HID devices
[2020-03-04 10:33:20.317119] [0x00007fa778dfc700] [error]   Error getting: `http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx`: HTTP 0, Unsupported protocol:
Protocol "https" not supported or disabled in libcurl
[Error 1]
```

bubnikv, 2020-03-04:
It looks like we bumped up version of libcurl, which breaks SSL support in PrusaSlicer on Linux starting with PrusaSlicer 2.2.0-alpha, therefore the config updates and new slicer version notifications are not downloaded.

It was correct in PrusaSlicer 2.2.1. We shall fix that. Thanks for heads up.

tamasmeszaros, 2020-03-06:
We've patched our build server, now the AppImages should be fine.

closed 2020-03-06 by tamasmeszaros