prusa3d/PrusaSlicer #3770: Implement gradual infill to correctly support roofs

nordurljosahvida, 2020-03-04:
### Version
2.2.0-rc but applies to all previous versions as well

### Operating system type + version

Ubuntu 18

### Behavior

In countless prints my MK3S detects collisions and re-homes [thus *always* shifting layers to the left or right from that height onwards] when printing roofs of a model on infills lower than 20%. That's because it's hard to print a good first roof layer above a sparse infill as it's all bridging over the infill itself.

Cura has already solved this issue with gradual infill, automatically doubling [several times if requested] the infill density just below roof layers:

https://ultimaker.com/en/resources/52670-infill [see gradual infill section]

Before you point this out, modifier meshes or height range modifiers do not help in many cases, as one would have to manually adjust infill density under each and every roof in the model, which is impossible in so many models such as the one I'm printing as I'm writing this [the Chernobyl reactor], where I've had several crash detections in the first 6mm of print. See attached pictures:

[image]
[image]
[image]
[image]
[image]

* _Steps needed to reproduce the problem_

Print anything with sparse infill and many small roofs without modifying any print settings.

* _Expected Results_

Infill can be triggered to [or automatically figures out to] double density [one or several times as layers go up] just below the roof layers when infill is sparse.

* _Actual Results_

see above images

_Is this a new feature request?_

Yes but already implemented in Cura

#### Project File (.3MF) where problem occurs

[031942 chernobyl_main_top-lg ---PLAX-PVAX 40 PK3SMM2S Nz A1.zip](https://github.com/prusa3d/PrusaSlicer/files/4287397/031942.chernobyl_main_top-lg.---PLAX-PVAX.40.PK3SMM2S.Nz.A1.zip)

bubnikv, 2020-11-20:
PrusaSlicer 2.3.0-alpha3 implements adaptive cubic support infill. Does it fulfill your needs?

nordurljosahvida, 2020-11-27:
Yes perfectly thanks!

closed 2020-11-27 by nordurljosahvida