prusa3d/PrusaSlicer #3772: Allow to disable raft under support structures.

Nandox7, 2020-03-04:
### Version
2.2.0+ but seen is previous as well

### Operating system type + version
OS independent, but using Windows version

### 3D printer brand / version + firmware version (if known)
MK3S - not model related

### Behavior
Current behavior is for the slicer to add a raft layer under all support structures.
While that is not problematic when touched the bed (most of the times unnecessary but not the main issue here) it will cause problems when applied over areas of the part.

Example:
[image]

Part above have a area where support will be required and what slicer does it also add a layer of raft over it.
[image]

Before adding the actual support.
[image]

While if it was the bed it could make sense to help with adhesion, in this case is unnecessary and what will happen is that it will fuse with the part itself and become really hard to remove.

Ideally the user would be given the option to disable raft under support allowing to prevent these cases.

_Is this a new feature request?_
Yes

bubnikv, 2020-03-11:
I appreciate three likes, but let's make the request clear even to me first:

Supports consist of a number of bottom interface layers, a number of sparse layers and a number of top interface layers. Bottom interface layers are built above top object surfaces, while top interface layers support bottom layers of an object. If a support is built directly over print bed, then the bottom interface layers are replaced with a single expanded 1st layer zig-zag.

Do I understand you correctly, that you object the dense support interface layer which are built above object top surfaces and which support sparse support towers? Currently you can disable these dense bottom interface layers by decreasing number of support interface layers to zero, but this disables top interface layers as well. Do I understand you correctly that your request would be fulfilled by making the number of top and bottom interface layers configurable independently?

Nandox7, 2020-03-11:
Just reviewed your comment and checked again on the application and yes that is correct.
Allow for independent control of the settings for interface layers between bottom and top.

bubnikv, 2020-03-12:
> Allow for independent control of the settings for interface layers between bottom and top.

Not just the number of interfaces, but for the separation distance as well. This is something we have discussed internally long time ago, but there is always something.

Nandox7, 2020-03-12:
Will this allow to disable the 1st layer zig-zag that is introduced when touching the print bed?

I see its potencial need if the support structures are small, but in the case of bigger support surfaces it becomes unnecessary. Having used other slicers that do not add it and not having problems of support structures getting loose or having poor adhesion.

SolidRick, 2020-03-23:
Yes BOTTOM support interface layers (dense support) to me are not useful in most cases, as the part itself is a good base for starting the support structure. They just make the support more difficult to remove. I propose to create a checkbox "Avoid bottom support interface layers" or similar

bubnikv, 2020-03-23:
>  I propose to create a checkbox "Avoid bottom support interface layers" or similar

If you search the issue tracker, this has been discussed recently, I think one week ago.

EccentricVamp, 2020-04-29:
Just wanted to give this a bump, and add to the discussion. It would also be nice to have a way to turn off the bed-support interface as well. I use brims quite often, and find myself wrestling with the over-zealous bed-support interface.
[image]
(with support)
[image]
(without support)

Henry3DP, 2020-06-02:
I just have to second this request. Please make an option to disable the single expanded 1st layer zig-zag pattern. 

This would allow to  adjust  the supports much better especially for small detailed models.

vanjan14, 2020-06-26:
Also bumping this request to modify and/or disable the first layer zig-zag under support structures.

Thank you!

ruebarb, 2020-07-14:
Bumping this request, printing PETG on a Powder coated textured PEI, these rafts have to be chiseled off the sheet, quickly destroying the powder coating on the sheet. Please add a simple option to disable them, thanks

DaleReitz, 2020-07-14:
Any way to make a SMOOTH support roof that touches the print so you don't have that nasty rough area on your print when you remove the support material?

fmvalero, 2020-07-16:
I also wanted to bump this request. I've never seen any sense in this raft under supports. It is annoying most of the time and troublesome when it is over a printed surface. It doesn't help at all. I would greatly appreciate an option to disable it.

flowalistik, 2020-09-22:
I would also like to disable the raft under support material. A clear example of when it's needed:

I'm using a Prusa Mini which has a 180x180mm bed size. I want to print a 175mm part which needs support material on the edges. The model is printable, but when you enable support material, it becomes unprintable because the support material raft makes the part +180mm.

dartrax, 2021-01-07:
> Bumping this request, printing PETG on a Powder coated textured PEI, these rafts have to be chiseled off the sheet, quickly destroying the powder coating on the sheet. Please add a simple option to disable them, thanks

THIS is exactly the problem I have. Please let us disable this!

Iqwertz, 2021-01-08:
Same Problem also exists when printing with flex!

batata004, 2021-02-14:
So how do I remove the zig-zag/raft from the support? It's too annoying and is completelly useless on 95% of cases, just wastes plastic, energy and printing time. This issue is almost 1 year old and no news about it? If it will not be implemented - FINE - but please let us know that this will not be fixed.

hoegge, 2021-02-20:
The raft layer should always be on the bed and not just under the part. Another reason for a good raft layer is that the current very sparse layer on the bed when using supports can be very hard to get off the bed, since it is a lot of small independent structures that do not come off as one. So seems like the current raft feature is broken maybe because it is never used much by Prusa (always printing square objects :-D )

hoegge, 2021-02-20:
You should have the option to create a raft layer at bottom only or under part only or both. Bottom only like in Cura: https://support.ultimaker.com/hc/en-us/articles/360012613519-Build-plate-adhesion-settings#:~:text=A%20raft%20adds%20a%20thick,in%20Ultimaker%20Cura%20are%20extensive.

batata004, 2021-02-20:
@hoegge that's a great complaing of mine. The raft under the support most of the times it much harder from remove from the heated bed than the part itself. Most of the times corners of my parts lift a little bit but the support never, it's rocky solid at the bed. That's pretty BAD! The support should be easy to remove, most of the times I end up scrapping the bed surface to remove support material of the raft. That's really a terrible approach.

It makes completely no sense making a raft like that that "stucks" so much to the heated bed. THE PURPOSE OF THE SUPPORT IS SUPPORTING THE PART! It's not to glue the part to the bed. If that was the case I would use RAFTS on my part. 

I am amazed how bad supports are on PS. Really bad. That's a shame because everything else that I used on PS is amazing well done, with great results, good printing times... But the support is so badly thought.

bubnikv, 2021-02-24:
Implemented with fcb714cd241f0847522b9aac57da6b62acdd0152

New parameters raft_first_layer_density and raft_first_layer_expansion
to influence the 1st layer of raft or support.

closed 2021-02-24 by bubnikv

bubnikv, 2021-03-19:
fixed with 6f3f362

batata004, 2021-04-30:
@bubnikv you said this problem was fixed (last reply from you) but I downloaded latest prusa right now (2.3.1), my support print is like this:

[image]

But my print is like this:

[image]

Unfortunatelly, PS is still adding raft - which I dont want! Is there any other configuration I need to change in order to avoid that huge raft around the support?

clearbucketLabs, 2021-05-02:
> @bubnikv you said this problem was fixed (last reply from you) but I downloaded latest prusa right now (2.3.1), my support print is like this:
>
> [image]
>
> But my print is like this:
>
> [image] [...]

it isn't in 2.3.1.... probably 2.3.2?

ruebarb, 2021-07-21:
This still hasn't been implemented in 2.3.2

ruebarb, 2021-07-21:
2.3.3 just came out, still not implemented

dartrax, 2021-07-21:
> fixed with [6f3f362](https://github.com/prusa3d/PrusaSlicer/commit/6f3f3624c54e3cf2b508ca4558f2357604318d81)

@bubnikv, I wonder was this forgotten or are there other reasons for delaying it, for example dependencies to other changes planned for a bigger/later release?

batata004, 2021-07-21:
I cant believe how terrible is support on the latest build. I cant believe how such a good software still has the worst support generation (even the manual) in the top 5 slicer.

damccull, 2021-07-22:
This plagues me too. I print without supports whenever remotely possible, or I use Cura to print with supports and rafts. Bring this!

hoegge, 2021-07-22:
Or at least to be able to set the height of the rafts, so it can be 2-3 layers and hence easier to remove from the bed

bubnikv, 2021-08-10:
It will be released with the upcoming PrusaSlicer 2.4.0-alpha1.

dartrax, 2021-08-19:
This is the mail system at host wipper.bitpalast.net.

I'm sorry to have to inform you that your message could not
be delivered to one or more recipients. It's attached below.

For further assistance, please send mail to postmaster.

If you do so, please include this problem report. You can
delete your own text from the attached returned message.

                   The mail system

***@***.***>: host gmail-smtp-in.l.google.com[173.194.76.26] said:
    550-5.7.1 [136.243.124.154      12] Our system has detected that this
    message is 550-5.7.1 likely unsolicited mail. To reduce the amount of spam
    sent to Gmail, 550-5.7.1 this message has been blocked. Please visit
    550-5.7.1  https://support.google.com/mail/?p=UnsolicitedMessageError 550
    5.7.1  for more information. v18si3451067wmh.59 - gsmtp (in reply to end of
    DATA command)

dartrax, 2021-08-19:
This is the mail system at host wipper.bitpalast.net.

I'm sorry to have to inform you that your message could not
be delivered to one or more recipients. It's attached below.

For further assistance, please send mail to postmaster.

If you do so, please include this problem report. You can
delete your own text from the attached returned message.

                   The mail system

***@***.***>: host gmail-smtp-in.l.google.com[173.194.76.27] said:
    550-5.7.1 [136.243.124.154      12] Our system has detected that this
    message is 550-5.7.1 likely unsolicited mail. To reduce the amount of spam
    sent to Gmail, 550-5.7.1 this message has been blocked. Please visit
    550-5.7.1  https://support.google.com/mail/?p=UnsolicitedMessageError 550
    5.7.1  for more information. f13si4153145wmc.98 - gsmtp (in reply to end of
    DATA command)

dartrax, 2021-08-19:
This is the mail system at host wipper.bitpalast.net.

I'm sorry to have to inform you that your message could not
be delivered to one or more recipients. It's attached below.

For further assistance, please send mail to postmaster.

If you do so, please include this problem report. You can
delete your own text from the attached returned message.

                   The mail system

***@***.***>: host gmail-smtp-in.l.google.com[173.194.76.26] said:
    550-5.7.1 [136.243.124.154      12] Our system has detected that this
    message is 550-5.7.1 likely unsolicited mail. To reduce the amount of spam
    sent to Gmail, 550-5.7.1 this message has been blocked. Please visit
    550-5.7.1  https://support.google.com/mail/?p=UnsolicitedMessageError 550
    5.7.1  for more information. y16si3653862wro.97 - gsmtp (in reply to end of
    DATA command)