prusa3d/PrusaSlicer #3777: PrusaSlicer crashes

basgoris, 2020-03-04:
### Prusa Version
Prusa Version 2.1.1+win64

### Operating system type + version
Windows 10 Pro version 1909

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior
Prusa slicer crashes when:
* Open the file and start the command to slice. 
  *  the crash happened after I introduced the range option. 
* 
[image]

_Is this a new feature request?_ NO

#### Project File (.3MF) where problem occurs: 
[RobertMechSet1_inclSupportboxes.zip](https://github.com/prusa3d/PrusaSlicer/files/4288256/RobertMechSet1_inclSupportboxes.zip)

bubnikv, 2020-03-04:
Please test with PrusaSlicer 2.2.0-rc.
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0-rc

basgoris, 2020-03-05:
Thank you Bubnikv, the prusaSlicer 2.2.0-rc does noet crash. 
Cheers, Bas

bubnikv, 2020-03-05:
thanks for heads up. Closing.

closed 2020-03-05 by bubnikv