prusa3d/PrusaSlicer #3779: Add a Brim Offset Option (New Feature Request)

sgano, 2020-03-05:
### Version
2.2.0-rc

### Operating system type + version
MacOS

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior

Is this a new feature request? YES

When adding a Brim it would be useful to have an offset value such that the brim can be removed more easily.   A small space like 0.05mm or 0.1mm between the brim and the part can still provide support to the model but make it's removal much easier.     This could be similar to the skirt "Distance from object" option.  

I have tried to create this by using a skirt with multiple loops and with a small distance offset value -- but for some model shapes (like an arc shaped lithophane) the skirt doesn't follow the shape of the model like a brim does (see attached image as an example).  

#### Project File (.3MF) where problem occurs
[image]

bubnikv, 2020-03-05:
Could you prototype your proposal in some other slicer? I mean, does any
other slicer do what you are proposing?

By the way, PrusaSlicer 2.2.0-rc already creates gap between object's first
layer and brim if you use the elephant foot compensation. This is a change
in regard to PrusaSlicer 2.2.1.

sgano, 2020-03-05:
Yes this is possible in Simplify3D.  In that slicer they actually combine their Skirt and Raft options into a single set of options.  This is because in the limit as the skirt offset approaches zero a skirt becomes a brim (or in other words a brim is just a special case of a skirt that is close to or touches the model).  However, this assumes that the skirt follows the outline of the first layers, which doesn't always seem to be the case for concave shapes in PrusaSlicer.   I have attached screenshots for the interface in Simplify3D as well as the sliced result.  

Thanks for the idea of using the elephant foot compensation to achieve this!  Though for general ease of use this not an obvious solution.  

[image]

[image]

bubnikv, 2020-03-06:
Thanks for the information on S3D.

Do you think that these 50um makes any difference? We can implement such
feature, it is quite trivial, but as a rule we would like to not implement
knobs and switches that only confuse the user and they are not of an actual
benefit.

sgano, 2020-03-08:
That is a good and very fair question.   I took some time today to experiment with this.  I used the elephant foot compensation value as a way to simulate a brim offset value.  As my current setup is a bit different than when I used the 0.05mm value (I have a MK3S now vs a MK2) and my live-Z is probably a bit lower relatively.      Anyway, I varied the value from 0 to 0.3mm offset, and I noticed that at 0 and 0.1mm the brim was basically welded to the part and wouldn't come off cleanly at all.  At 0.2mm this improved and at 0.3mm it was even better. I still had to do some sanding to make it smooth but it seemed to still provide support to the foot of the model (tree frog) but allow the brim to be "scored" so it would tear away easier and cleaner.  This definitely was not a exhaustive test.  But it did seem to make a real difference and I would think that having a brim offset value would be at least useful in the "advanced" mode.   

[image]

[image]

[image]

sgano, 2020-03-08:
(Default offset values would be tough to set though, as this really depends on each person's live-z value and how much first layer "squish" they have/prefer.)  I would suggest in the 0.2mm range for PLA.

bubnikv, 2020-03-08:
PrusaSlicer 2.2.0 has 0.2mm elephant foot compensation set in Prusa's
system profiles by default.

I understand that it may be nice to be able to remove the brim without
sanding, but then I wonder about the functionality of such a brim. The brim
is there to improve bed adhesion especially for materials that shrink
considerably, like ABS or nylon. I am afraid that if you increase your gap
between the object and the brim, the brim will no longer serve its purpose.

sgano, 2020-03-08:
Upon thinking of this a few more notes to share:

1) I used the TreeFrog model as the test because I do generally have issues printing the front legs without a brim in place.  So the small gap in the brim does provide support -- with a small gap it still is continuous plastic -- there is just a thinner line around the part which makes it easier to remove as it is "pre-scored" -- so there is still lateral support just not as much support in and out of the plane of the bed. 

2) The elephant's foot compensation is actually "backwards" from a brim offset value as it under extrudes the first layer dimension -- an offset for the brim would be offset on the outside -- which is better as it is less likely to weld to the layer above.   So my simulation was not quite what should be expected with a real offset.

3) If you do decide to add the feature setting the offset to 0 by default does make a lot more sense as that would essentially default back to the original behavior when adding a brim.

4) I didn't invent this technique, it is used quite extensively in the Simplify3D community.   I learned it from watching tutorial videos on YouTube.   You can also find people mentioning doing this all over their forum and it is even recommended in their own documentation:

A) A small offset for the brim is even recommended in their documentation to make it easier to separate: https://www.simplify3d.com/support/articles/rafts-skirts-and-brims/

B) I found another user asking for it in PrusaSlicer - but in the PrusaPrinters forums not on GitHub:  https://forum.prusaprinters.org/forum/prusaslicer/option-for-brim-offset/?mode=grid&forum=377&topic=33668

C) Other references to people doing this:
https://forum.simplify3d.com/viewtopic.php?t=7632
https://forum.simplify3d.com/viewtopic.php?t=2900

bubnikv, 2020-03-09:
BTW what material are you printing the tree frog with and on which print
bed?
If you are squishing the first layer sufficiently, you should want the
elephant foot compensation as the 1st layer will be about 0.2mm wider than
it should be.

sgano, 2020-03-09:
I am using PLA (specifically: FilamentOne PLA PRO SELECT Glint Evergreen) for the Frog.  Printed on the Smooth PEI sheet.   (Yes the elephant foot compensation definitely helps especially on parts that have tight tolerances; it is a great feature in its own right).

Nandox7, 2020-04-10:
+1 to have the same options (or at least some) as for  skirt.
I'm used (in other slicers) to give a small offset (+-0.1mm ) to make it easier to peel off the brim.
Without it becomes much harder and most of the times I have to use a blade to cut leaving a less than ideal finish.
Alternative so fa was to sue skirt, but depending on the part of if ti has support it not great as it would fully touch the part.

stalawina, 2020-04-22:
Oh yes I really want this to! I do really long prints in Nylon. Like 50 hrs prints. The only effective way I have to prevent warping is to add a Brim with 0 mm offset from the part. I can then easily cut the brim. I can do this easily with S3D, but not in PrusaSlicer.
PrusaSlicer has really neat features that make me want to switch to it and S3D obviously will not get updated, but the non-functionnal brim function ins PrusaSlicer prevent me from moving to it.
Thanks!

johnbiggs, 2020-04-23:
Actually issue 3826 might be solved with a negative offset for that user. So an offset option could work for a few scenarios.

bubnikv, 2021-02-24:
Implemented.

closed 2021-02-24 by bubnikv

schornhe, 2021-05-23:
How is this implemented? I have just installed the latest update PrusaSlicer 2.3.1 (win), but I don't see any option in expert mode related to brim.

linuxtextadventurer, 2021-07-09:
> 
> 
> Implemented.

I'm running `Version 2.3.2+win64` and can't find the feature requested. Can you please mention the commit in which this was implemented?

rtyr, 2021-08-12:
It is implemented on master. It will be included in the upcoming PrusaSlicer 2.4.0-alpha1.