prusa3d/PrusaSlicer #3780: PrusaSlicer draft protection error.

Wieslaw-Andrzej, 2020-03-05:
I think I found a bug in PrusaSlicer from version 2.2.0-alpha4 + win32 / win64 to the latest published version.
Up to version PrusaSlicer 2.2.0-alpha2 the draft shield worked correctly. It allowed to create a protective wall 1 (one perimeter). Width in the picture is better to explain.

[image]

[image]

From PrusaSlicer 2.2.0-alpha4 version, the draft shield is working badly. Does not allow the creation of a protective wall 1 (one perimeter). It's better to explain in the picture.

[image]

[image]

This feature is very useful for printing from ABS filaments on open printers. It is a pity that it does not work properly.

Regards

bubnikv, 2020-03-05:
The "minimal extrusion length" value you entered seems fishy to me. It
looks like the number is newly respected by PrusaSlicer 2.2.0, therefore
more than a 1 loop is extruded. Indeed, the "number of loops (minimum)"
indicates than more than a single loop may potentially be extruded.

bubnikv, 2020-03-14:
No response, closing.

closed 2020-03-14 by bubnikv