prusa3d/PrusaSlicer #3781: 2.2.0-rc cmake unit test fail and do not terminate on 32bit

bnavigator, 2020-03-05:
I get the following output after compiling 2.2.0-rc and executing the cmake unit tests:

```
[ 1306s] Executing(%check): /bin/sh -e /var/tmp/rpm-tmp.Ia2Foe
[ 1306s] + umask 022
[ 1306s] + cd /home/abuild/rpmbuild/BUILD
[ 1306s] + cd PrusaSlicer-version_2.2.0-rc
[ 1306s] + cd build
[ 1306s] ++ [[ -n '' ]]
[ 1306s] +++ /usr/bin/getconf _NPROCESSORS_ONLN
[ 1306s] ++ echo -j8
[ 1306s] + /usr/bin/ctest --output-on-failure --force-new-ctest-process -j8
[ 1306s] Test project /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-rc/build
[ 1306s]     Start 1: libnest2d_tests
[ 1306s]     Start 2: libslic3r_tests
[ 1306s]     Start 3: timeutils_tests
[ 1306s]     Start 4: fff_print_tests
[ 1306s]     Start 5: sla_print_tests
[ 1306s] 1/5 Test #3: timeutils_tests ..................   Passed    0.01 sec
[ 1306s] 2/5 Test #1: libnest2d_tests ..................***Failed    0.49 sec
[ 1306s] Testing Angles
[ 1306s] Passed in 0.00098 [seconds]
[ 1306s] 
[ 1306s] Testing ItemCreationAndDestruction
[ 1306s] Passed in 3e-05 [seconds]
[ 1306s] 
[ 1306s] Testing boundingCircle
[ 1306s] Passed in 0.001482 [seconds]
[ 1306s] 
[ 1306s] Testing Distance
[ 1306s] Passed in 9.2e-05 [seconds]
[ 1306s] 
[ 1306s] Testing Area
[ 1306s] Passed in 3.8e-05 [seconds]
[ 1306s] 
[ 1306s] Testing IsPointInsidePolygon
[ 1306s] Passed in 2.3e-05 [seconds]
[ 1306s] 
[ 1306s] Testing LeftAndDownPolygon
[ 1306s] Passed in 8.5e-05 [seconds]
[ 1306s] 
[ 1306s] Testing ArrangeRectanglesTight
[ 1306s] Passed in 0.004457 [seconds]
[ 1306s] 
[ 1306s] Testing ArrangeRectanglesLoose
[ 1306s] Passed in 0.004601 [seconds]
[ 1306s] 
[ 1306s] Testing BottomLeftStressTest
[ 1306s] Passed in 0.002563 [seconds]
[ 1306s] 
[ 1306s] Testing convexHull
[ 1306s] Passed in 2.5e-05 [seconds]
[ 1306s] 
[ 1306s] Testing PrusaPartsShouldFitIntoTwoBins
[ 1306s] Passed in 0.32181 [seconds]
[ 1306s] 
[ 1306s] Testing EmptyItemShouldBeUntouched
[ 1306s] Passed in 3.1e-05 [seconds]
[ 1306s] 
[ 1306s] Testing LargeItemShouldBeUntouched
[ 1306s] Passed in 2.4e-05 [seconds]
[ 1306s] 
[ 1306s] Testing Items can be preloaded
[ 1306s] Preloaded Item should be untouched
[ 1306s] Preloaded Item should not affect free bins
[ 1306s] Passed in 0.002114 [seconds]
[ 1306s] 
[ 1306s] Testing nfpConvexConvex
[ 1306s] Passed in 0.002803 [seconds]
[ 1306s] 
[ 1306s] Testing pointOnPolygonContour
[ 1306s] 
[ 1306s] ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 1306s] libnest2d_tests is a Catch v2.9.2 host application.
[ 1306s] Run with -? for options
[ 1306s] 
[ 1306s] -------------------------------------------------------------------------------
[ 1306s] pointOnPolygonContour
[ 1306s] -------------------------------------------------------------------------------
[ 1306s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-rc/tests/libnest2d/libnest2d_tests_main.cpp:975
[ 1306s] ...............................................................................
[ 1306s] 
[ 1306s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-rc/tests/libnest2d/libnest2d_tests_main.cpp:992: FAILED:
[ 1306s]   REQUIRE( shapelike::touches(v, input.transformedShape()) )
[ 1306s] with expansion:
[ 1306s]   false
[ 1306s] 
[ 1306s] Testing mergePileWithPolygon
[ 1306s] Passed in 4.5e-05 [seconds]
[ 1306s] 
[ 1306s] Testing MinAreaBBWithRotatingCalipers
[ 1306s] Passed in 0.132752 [seconds]
[ 1306s] 
[ 1306s] ===============================================================================
[ 1306s] test cases:   19 |   18 passed | 1 failed
[ 1306s] assertions: 1746 | 1745 passed | 1 failed
[ 1306s] 
[ 1306s] 
[ 1314s] 3/5 Test #2: libslic3r_tests ..................   Passed    8.34 sec
[ 1315s] 4/5 Test #4: fff_print_tests ..................   Passed    9.57 sec
```
`libnest2d_tests::pointOnPolygonContour()` fails and `sla_print_tests()` is started but never terminates. The build vm is then terminated after 30000s of no response.

Any idea how I can make the tests pass? I would rather not skip them in the build recipe for the package as this will likely not be accepted by the OpenSUSE repository managers.

tamasmeszaros, 2020-03-05:
Hi @bnavigator! Can you tell what version of OpenSUSE and boost libraries are you using for the build? I assume you are building with dependencies installed from package manager, am I right?

bnavigator, 2020-03-05:
Yes standard OpenSUSE Tumbleweed packages on the OpenSUSE build system.

You can see the specfiles and buildlogs here:
https://build.opensuse.org/package/show/home:bnavigator:branches:science/PrusaSlicer

Currently the build does not execute the ctest because my attempt at limiting the number of parallel processes failed with a wrong working directory. Will fix that later

bnavigator, 2020-03-05:
Boost version is 1.71.0, the packages are:

```specfile
BuildRequires:  libboost_atomic-devel
BuildRequires:  libboost_filesystem-devel
BuildRequires:  libboost_iostreams-devel
BuildRequires:  libboost_locale-devel
BuildRequires:  libboost_log-devel
BuildRequires:  libboost_regex-devel
BuildRequires:  libboost_system-devel
BuildRequires:  libboost_thread-devel
```

For completeness, I should mention that 64bit builds fine.

bnavigator, 2020-03-06:
Here is a new build log with `ctest -V` output: https://github.com/prusa3d/PrusaSlicer/files/4298481/_log.txt

```
[ 1864s] + ctest -V --output-on-failure --force-new-ctest-process -j1 --timeout 600
...
[ 1864s] 1: Testing pointOnPolygonContour
[ 1864s] 1: 
[ 1864s] 1: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 1864s] 1: libnest2d_tests is a Catch v2.9.2 host application.
[ 1864s] 1: Run with -? for options
[ 1864s] 1: 
[ 1864s] 1: -------------------------------------------------------------------------------
[ 1864s] 1: pointOnPolygonContour
[ 1864s] 1: -------------------------------------------------------------------------------
[ 1864s] 1: /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-rc/tests/libnest2d/libnest2d_tests_main.cpp:975
[ 1864s] 1: ...............................................................................
[ 1864s] 1: 
[ 1864s] 1: /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-rc/tests/libnest2d/libnest2d_tests_main.cpp:992: FAILED:
[ 1864s] 1:   REQUIRE( shapelike::touches(v, input.transformedShape()) )
[ 1864s] 1: with expansion:
[ 1864s] 1:   false
[ 1864s] 1: 
...
[ 1885s]     Start 5: sla_print_tests
[ 1885s] 
[ 1885s] 5: Test command: /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-rc/build/tests/sla_print/sla_print_tests
[ 1885s] 5: Test timeout computed to be: 600
[ 1885s] 5: Testing Raycaster - find intersections of a line and cylinder
[ 1885s] 5: Passed in 2.5e-05 [seconds]
[ 1885s] 5: 
[ 1885s] 5: Testing Raycaster with loaded drillholes
[ 1889s] 5: 
[ 1889s] 5: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 1889s] 5: sla_print_tests is a Catch v2.9.2 host application.
[ 1889s] 5: Run with -? for options
[ 1889s] 5: 
[ 1889s] 5: -------------------------------------------------------------------------------
[ 1889s] 5: Raycaster with loaded drillholes
[ 1889s] 5: -------------------------------------------------------------------------------
[ 1889s] 5: /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-rc/tests/sla_print/sla_raycast_tests.cpp:46
[ 1889s] 5: ...............................................................................
[ 1889s] 5: 
[ 1889s] 5: /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-rc/tests/sla_print/sla_test_utils.cpp:286: FAILED:
[ 1889s] 5:   REQUIRE_FALSE( mesh.needed_repair() )
[ 1889s] 5: with expansion:
[ 1889s] 5:   !true
[ 1889s] 5: 
[ 1889s] 5: Testing Pillar pairhash should be unique
[ 1889s] 5: Passed in 0.002695 [seconds]
[ 1889s] 5: 
[ 1889s] 5: Testing Support point generator should be deterministic if seeded
[ 1889s] 5: Passed in 0.257946 [seconds]
[ 1889s] 5: 
[ 1889s] 5: Testing Flat pad geometry is valid
[ 1889s] 5: Passed in 0.005524 [seconds]
[ 1889s] 5: 
[ 1889s] 5: Testing WingedPadGeometryIsValid
[ 2485s] 5/5 Test #5: sla_print_tests ..................***Timeout 600.11 sec
[ 2485s] Testing Raycaster - find intersections of a line and cylinder
[ 2485s] Passed in 2.5e-05 [seconds]
[ 2485s] 
[ 2485s] Testing Raycaster with loaded drillholes
[ 2485s] 
[ 2485s] ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 2485s] sla_print_tests is a Catch v2.9.2 host application.
[ 2485s] Run with -? for options
[ 2485s] 
[ 2485s] -------------------------------------------------------------------------------
[ 2485s] Raycaster with loaded drillholes
[ 2485s] -------------------------------------------------------------------------------
[ 2485s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-rc/tests/sla_print/sla_raycast_tests.cpp:46
[ 2485s] ...............................................................................
[ 2485s] 
[ 2485s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-rc/tests/sla_print/sla_test_utils.cpp:286: FAILED:
[ 2485s]   REQUIRE_FALSE( mesh.needed_repair() )
[ 2485s] with expansion:
[ 2485s]   !true
[ 2485s] 
[ 2485s] Testing Pillar pairhash should be unique
[ 2485s] Passed in 0.002695 [seconds]
[ 2485s] 
[ 2485s] Testing Support point generator should be deterministic if seeded
[ 2485s] Passed in 0.257946 [seconds]
[ 2485s] 
[ 2485s] Testing Flat pad geometry is valid
[ 2485s] Passed in 0.005524 [seconds]
[ 2485s] 
[ 2485s] Testing WingedPadGeometryIsValid
[ 2485s] 
[ 2485s] 
[ 2485s] 60% tests passed, 2 tests failed out of 5
[ 2485s] 
[ 2485s] Total Test time (real) = 620.84 sec
[ 2485s] 
[ 2485s] The following tests FAILED:
[ 2485s] 	  1 - libnest2d_tests (Failed)
[ 2485s] 	  5 - sla_print_tests (Timeout)
[ 2485s] Errors while running CTest
[ 2485s] error: Bad exit status from /var/tmp/rpm-tmp.e69Zuq (%check)

```

tamasmeszaros, 2020-03-06:
@bnavigator I see there are more tests failing. I will dive into this problem as soon as I can.

tamasmeszaros, 2020-03-12:
Hi @bnavigator,

after investigation, we've found that all tests run successfully if you include either the `-ffloat-store` or `-mfpmath=sse -msse2` into C and CXX flags for compilation. It depends on what target architecture you want to support. If i586 is the target (as the Tubleweed image is labeled) I would say that `-ffloat-store` is the way to go. See the corresponding article: https://gcc.gnu.org/wiki/FloatingPointMath (_Section: Note on x86 and m68080 floating-point math_)

Good luck!

bnavigator, 2020-03-12:
Awesome! I tried both options and they both build successfully.

I could not find a definition what platform the Tumbleweed i586 packages have to support. I don't know whether anyone would want to run PrusaSlicer on a Tumbleweed machine with a Pentium III or older. So I guess I will submit the package for i586 with `-mfpmath=sse -msse2` to OpenSUSE:Factory as soon as PrusaSlicer 2.2.0 is released. The reviewers will tell me if those flags are not acceptable.

closed 2020-03-12 by bnavigator