prusa3d/PrusaSlicer #3782: Turning off "Use free camera" requires application restart to take effect

mikolaszuza, 2020-03-05:
PrusaSlicer 2.2.0 RC1+17

Turning off **Use free camera** requires PrusaSlicer restart to take effect.

When the user turns it off, it would be nice to reset the view matrix to some default value and turn it off, so that it's not necessary to restart the application.

foreachthing, 2020-03-05:
How many times, per session, do you switch back and forth?

mikolaszuza, 2020-03-05:
I don't. And I don't think it's necessary to switch back and forth.
I think it might be a bit confusing for the user, who discovers this feature in the preferences, tries it out, realizes it's not what he wants, turns it off and realizes nothing happens and the camera still behaves like it's on.

It's an absolutely low priority issue. However, the solution should be pretty straight forward, so I thought I might as well create an issue for it.

bubnikv, 2020-03-06:
I think this is a result of my refactoring after 2.2.0-rc

bubnikv, 2020-03-07:
: I think this is a result of my refactoring after 2.2.0-rc

So actually the issue is after switching from free to constrained camera, the up vector of the camera should be aligned with Y axis, which it is not.

bubnikv, 2020-03-11:
Fixed in prusaSlicer 2.2.0-rc3. Closing.

closed 2020-03-11 by bubnikv