prusa3d/PrusaSlicer #3783: Some vertical dimensions are wrong

rennne, 2020-03-05:
When I design a part with a square hole of 2.0mm height. Slicer will produce a part with about 1.8mm height.

It is even shown in the layer-preview, that the resulting air gap is only 9 layer instead of 10 at 0.2mm layer height:
[image]

The preview show that die bridge will take up two layers in the vertical dimension.

So may conclusion would be, that bridges are placed one layer too low.

BMacours, 2020-11-05:
I will follow this one. Seems to be linked with [https://github.com/prusa3d/PrusaSlicer/issues/3458](url)
Its very annoying when building precision parts. Prusa Slicer should have the ability to compensate bridging and overhanging heights. Thanks in advance to Prusa devs!

Equidamoid, 2020-11-19:
Same problem. 
I have a model which is essentially a "T" rotated 90 degrees (like ⊢, with really good reasons to make it that way).

Interesting thing is that if I remove the overhang layer, the dimensions are almost perfect. Happens for both 0.2 and 0.1mm layers, so probably not a coincidence.
Removing it is a pain though, especially perimeters. Also it requires some filing/sanding afterwards, which is evil.

BMacours, 2020-11-19:
Link updated ;-)

closed 2026-09-29 by prusaslicer-bot[bot]