prusa3d/PrusaSlicer #3785: Ignoring fan speed settings

jan-b45, 2020-03-05:
### Version
2.1.1

### Operating system type + version
WIN10 x64

### 3D printer brand / version + firmware version (if known)
MK2.5S MMU2S

### Behavior
I am using settings to decrease cooling fan speed to 80%. When I used this setting on MK2.5S without MMU2S (Slicer profile MK2.5S), it works fine. After printer upgrade to MMU2S (Slicer Profile MK2.5S MMU2S) is this setting ignored. 
When I open generated g-code and look at the M106 value, it is always 255 doesnt matter what was set in Slicer and it is not printer FW fault.

[image]

bubnikv, 2020-03-07:
>  I am using settings to decrease cooling fan speed to 80%.

What settings are you using?
Please provide a 3MF.

jan-b45, 2020-03-07:
I am using settings for MK2.5S  MMU2S and filament profile default PrusaPLA only with fan speed changes. Here is the project file 
[OtvirakBilyText.zip](https://github.com/prusa3d/PrusaSlicer/files/4302053/OtvirakBilyText.zip)

rtyr, 2020-03-08:
Are you sure that this is the correct project file? I tried to open it and there is stock "Prusament PLA MMU2" filament profile selected (with default 100% fan speed). I also tried to change values according to your screenshot and I was unable to reproduce issue.

jan-b45, 2020-03-08:
Issue is, that I am not able to save (or export) file with different than 100% fan speed. I changed fan speed, clic save, close the Slicer. Open Slicer again, open previousl saved file, look at the fan speed settings and see 100% fan speed. Same situation is when I export g-code.

rtyr, 2020-03-08:
Make sure, that you have proper filament profile selected here (you have to do it after changing filament profile directly in the "filament settings tab").

[image]

rtyr, 2020-03-08:
This is how it should be set up if you want different settings for extruder 3 and 4.

[OtvirakBilyText_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/4303737/OtvirakBilyText_fixed.zip)

jan-b45, 2020-03-09:
@rtyr Thanks for the help, this was my error, I changed speeds globaly, but no per filament. I tried to change it per filament and settings is used for printing. Thanks again

closed 2020-03-09 by jan-b45