prusa3d/PrusaSlicer #3786: Feature request: Allow copying modifiers from one object to another

mdaneman, 2020-03-05:
Sometimes I'm printing mutiple similar (but not identical) objects at once and, if I'm using modifier objects (support blockers/enforcers, other modifiers), it would be very useful if I could copy the modifer objects from one main object to another.  In some cases it takes a fair amount of time to adjust the modifier objects' size and positions, and it would save time to be able to copy them to another main object that has similar features.

lukasmatena, 2020-03-05:
It should be already possible in 2.1.1 already. Just select the modifier, Ctrl+C, select the other object in the list, Ctrl+V. Does that not work for you?

mdaneman, 2020-03-05:
Holy crap, it works!  I could have sworn that I've tried this multiple times before and didn't have it working.  Maybe this is something that was implemented in the recent version?
In any case, thanks a lot.  This issue can be closed.

closed 2020-03-05 by mdaneman