prusa3d/PrusaSlicer #3787: Is it possible for holes to have more walls?

FlailAway, 2020-03-05:
### Version
2.xx

### Operating system type + version
Any

### 3D printer brand / version + firmware version (if known)
Any

### Behavior
Be able to define holes to have more walls than the main body. I need to tap some holes so I have set wall count to "5" but this prints 5 walls for everything and they are really not needed. It adds time and wastes PLA.

Is there some way to define all holes (regardless of axis orientation) to have more walls?

I don't mean individually select and "fatten" at slicing time, but as an option for all holes. I can live with a few extra non-tap holes getting walls they do not need, but it would be better than the entire print having 5-walls.

neophyl, 2020-03-05:
How does the slicer determine a hole from another interior feature ?  The usual method is to place modifier meshes in the places you want more perimeters and then set them up to have whatever you decide is needed for tapping, 5 in this case.

It would be great if it could magically determine holes though no matter their placement and orientation as it would save me a bunch of time placing modifier meshes :)

FlailAway, 2020-03-05:
> How does the slicer determine a hole from another interior feature ?

I would imagine it would be similar to identifying horizontal holes requiring support within the hole. Seems to me, that process discovers a hole just fine.

A hole for tapping is unique in that it is A: Round, B: Hollow for some length.

I would be happy to define the holes by clicking on an edge to be treated as a tapping hole and thus, give it extra walls. Yes, it would have to be done every slice, but way more easily so than than the Mess, ooops, Mesh process.

As to Mesh, I covered that in the OP as an undesirable option as it must be done for every hole in every slice. That' gets tedious fast.

The idea is to try and help make something new happen rather than stonewall.

bubnikv, 2021-06-03:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3846

closed 2021-06-03 by bubnikv