prusa3d/PrusaSlicer #3791: 2.2.0 RC - 3Dconnexion settings not saved

dahuafschmied, 2020-03-07:
### Version
2.2.0 RC

### Operating system type + version
MacOS 10.13.6

### Behavior
Since the RC the 3Dconnexion settings (cmd+m) are not stored anymore. Close and Reopen the Prusaslicer resets them to default values. (by the way: the default values are way off as i am used to in "normal" CAD programs and the zoom is binary not analog)

[image]

bubnikv, 2020-03-07:
This piece has been reworked in PrusaSlicer 2.2.0-rc2 to be released today,
it should work.
Please retest and close this issue if it is fixed in PrusaSlicer 2.2.0-rc2.
Thank you.

dahuafschmied, 2020-03-08:
Sorry not fixed.
Values are still not stored and zoom is still digial

bubnikv, 2020-03-08:
> by the way: the default values are way off as i am used to in "normal" CAD programs and the zoom is binary not analog

Could you provide a screenshot with the sensitivities that you are used to? Maybe you want to set the deadzone to zero? 

> the zoom is binary not analog

What do you mean that? You mean zooming with pulling the knob towards you and away from you?

bubnikv, 2020-03-09:
> update: set the speeds to 10 is perfect. and deadzone to 0. (zoom should
> be at 10 to. but you have to fix the on/off bug first)
> [image: Bildschirmfoto 2020-03-08 um 10 58 20]
> <https://user-images.githubusercontent.com/3955019/76160593-c3854300-612b-11ea-82d6-415b5a0036f1.png>

I can confirm that the settings is not saved, but I cannot reproduce the
binary zoom. Do you still have it? Maybe your settings in the system driver
is set to extremes? What happens if you set the resolution in 3Dconnexion
driver to default and then set the sensitivity in slicer?

dahuafschmied, 2020-03-09:
i already reseted the mouse settings to default values.
[image]

zoom is still binary.

i changed the values for testing but it stays binary...

dahuafschmied, 2020-03-09:
in Fusion360 the zoom works as it should

bubnikv, 2020-03-10:
Would you make a movie of the binary zoom please? We are not able to
reproduce it in any way.

dahuafschmied, 2020-03-10:
this last commit seems logical to me.
is it possible to get a test version? or do i have to compile it by myself?

enricoturri1966, 2020-03-11:
@dahuafschmied 
The fix will be part of next release candidate, which should be published later today.

closed 2020-03-11 by bubnikv

jweaver100, 2020-03-11:
After trying 2.2.0 out last night, i immediatly noticed the zoom issue you described.. Where as all of the other axis are 'analogue' (i.e the amount you move the puck dictates the speed of travel), the zoom is either on or off.. So you have no 'analogue' control over it.

Are we saying that the fix thats been suggested/implimented is going to address this?

dahuafschmied, 2020-03-11:
3D Mouse is perfect now! thank you for the fast and helpful support!

bubnikv, 2020-03-12:
Great. Thanks for letting us know.
You would not believe how time consuming thus expensive the whole
3DConnexion integration was.

foreachthing, 2020-03-12:
> You would not believe how time consuming thus expensive the whole 3DConnexion integration was.

I appreciate every second you invested in this feature!!! :-D

Now ... let's talk about function keys ... _duckandrun_

dahuafschmied, 2020-03-12:
> Now ... let's talk about function keys ... _duckandrun_

Buttons on 3DMouse with PrusaSlicer:
[image]
[image]

Works Perfect

foreachthing, 2020-03-12:
Yes, I started using _Macros_, too. But if PS could use the default keys ... that'd be something!!! :-)
[image]
Now, you have to program **all** the functions (macros) manually - basically make a macro for each PS-function you want to map. Not impossible, just tedious.

jweaver100, 2020-03-12:
I am not sure I understand what the issue is with buttons.. I have an unsupported SpacePilot which works well.. And whilst the speed/sensitives don't work in the 3DConnexion setup tool (but CTL+M works in Slic3r), the button seem fine.

So I have been able to add all the features I want assuming that Prusa already has hotkeys for the function.

For me, 3D mouse support is working very well... Its just a shame that having one connected affects the way that a "normal" mouse works.

foreachthing, 2020-03-12:
> I am not sure I understand what the issue is with buttons.. 

When I click the TOP (-view)-Button (SpacePilot Pro), it does nothing. I have to assign an custom macro (where I have to map the [1]-key to) to this button. Short: create a macro where I map [1] to. Assign this macro to the TOP-button.... whereas SolidWorks just takes the TOP-botton command and shows the top view. Same for front, back, etc.
This is, however, a topic for a new issue.

bubnikv, 2020-03-12:
It would be great if you guys could export your 3DConnexion button
assignments from your 3DConnexion settings dialog and share it somewhere
for the others. One will not fit all as there are so many different
3DConnexion models with different number of buttons. I believe this is a
better way than trying to integrate it into our application.

foreachthing, 2020-03-16:
Here are my settings for my SpacePilot:
https://github.com/foreachthing/PrusaSlicer-3DConnexionSettings

bubnikv, 2020-03-16:
thanks, I have passed it to our content guys for the manual.