prusa3d/PrusaSlicer #3792: PS 2.2 RC1 Perimeter printed two times per layer

fdurrer, 2020-03-07:
### Version
PrusaSlicer 2.2.0 RC1

### Operating system type + version
Win 10 pro 64bit

### 3D printer brand / version + firmware version (if known)
mk3S

### Behavior
When slicing the attached 3mf file, PS 2.2.0 generates the gcode of certain perimeters two times. I realized this when I was watching the first layer getting printed.
When I slice the same 3mf file with PS 2.1.1 the error does not occur.
I tried to compare the two gcode files, but as the sequence of 
[Distanzstuecke_3x.zip](https://github.com/prusa3d/PrusaSlicer/files/4301770/Distanzstuecke_3x.zip)

objects is not the same, it's quite impossible. In the gcode file generated by PS 2.2.0 you can find the duplicated gcode (in the attached gcode files its fir the first layer line 34 and the duplicate starts on line 332)

To reproduce:
slice attached 3mf file with PS 2.2.0 RC1 analyze the gcode with http://gcode.ws/ and check the first layer of the first object

bubnikv, 2020-03-07:
I cannot reproduce what you have described. I reviewed the G-code in S3D G-code viewer and I don't see any issue.

Could you make screenshots of the gcode.ws page where it happens? Where is the layer line number shown at gcode.ws? I cannot find it.

Why do you think that there is such an issue in PrusaSlicer in the first place?

I remember seeing such an issue being reported for multi-material prints, but it is very rare and we were not able to reproduce it in house, or maybe we were, but it was extremely rare, so we don't know why it happens. We believe though that this issue is caused by MM wipe to infill / wipe to another object features.

bubnikv, 2020-03-07:
@rtyr claims that he sees your issue, but when he generated the G-code the G-code is fine.

bubnikv, 2020-03-07:
Now I see that the G-code that you provided generated by PrusaSlicer 2.2.0 really contains duplicate perimeters of the front row of objects, though the G-code I generated from your 3MF is fine. Do you have an idea how to reproduce the issue?

lukasmatena, 2020-03-07:
The other, extremely rare issue that @bubnikv mentioned is #2540 (although the discussion is now cluttered by probably unrelated issues). I have a list of steps to reproduce it, but it really only happens with MM prints, at least as far as I know. I did not find the cause when I was looking into it.

However, it might be possible that the issue can actually manifest with single extruder prints as well and that it is in fact what happened in your case. I don't know if wipe into infill is responsible, but I managed to provoke it without even turning that feature on. If you found a way to provoke it again, it would help. But it may be very difficult to do.

fdurrer, 2020-03-07:
> Now I see that the G-code that you provided generated by PrusaSlicer 2.2.0 really contains duplicate perimeters of the front row of objects, though the G-code I generated from your 3MF is fine. 

Yes. I tried to make screenshots from gcode.ws but you can't see anything as te lines are on each other. You can best see it in the gcode itself.

> Do you have an idea how to reproduce the issue?
Honestly no, but I will try. I have defined a dedicated infill for both parts on the plater. Thats the only special setting I'm aware of.

fdurrer, 2020-03-07:
Sorry, I'm unable to reproduce it.
When I delete the wrongly sliced part, add it new and apply the same settings, PS 2.2.0 slices everything correctly.
I will report here, when I'm able to reproduce.
Thanks for your effort guys!

lukasmatena, 2020-10-16:
@fdurrer Did you manage to reproduce the issue? it is very likely that it had the same cause as #2540, which was fixed for 2.2.0 final.

rtyr, 2021-06-08:
No response. Should be fixed. Closing.

closed 2021-06-08 by rtyr