prusa3d/PrusaSlicer #3793: Unable to eject SD card on macOS

aequitas, 2020-03-07:
### Version
2.2.0-rc2

### Operating system type + version
macOS 10.14.6

### Behavior
When inserting a SD card into the SD slot of my iMac PrusaSlicer does not recognise it and does not show any buttons to eject it. Also the menu shortcut is grayed out.

Clicking "Export G-code" button does show the SD card in the list of devices on the filebrowser and slicer is able to save the file there.

Is SD Card handling a Windows only feature?

bubnikv, 2020-03-08:
Does  2.2.0-rc works for you in this regard? Will it detect both USB and SD
card plug / unplug events?

We have switched from polling to operating system notifications on
2.2.0-rc2 on OSX. On other platforms we are still polling, now at a
background thread to not block the user interface. We are
receiving NSWorkspaceDidMountNotification
/ NSWorkspaceDidUnountNotification events from OSX.

We have found on Friday, that these events do not come for USB flash drive,
which is marked as bootable. Maybe there is another weirdness to this OSX
notification API that we are not aware of? Would you please try more SD
cards?

bubnikv, 2020-03-08:
If we discover more weirdness to the OSX mount / unmount notification API,
we may switch back to polling.

aequitas, 2020-03-08:
It didn't work in any version before, not even the version which introduced this feature.

I'll try to get a build environment up and debug if the event is received when inserting the SD cardand calls the function here https://github.com/prusa3d/PrusaSlicer/blob/b3b800de656db89d0b50ca574b15eea0796383f7/src/slic3r/GUI/RemovableDriveManagerMM.mm#L21. We can then continue debugging from there.

brettvitaz, 2020-03-09:
I can verify that eject does work for MacOS on my machine, but it is very slow. This is true for me for all versions since introduction.

bubnikv, 2020-03-09:
>  I can verify that eject does work for MacOS on my machine, but it is
very slow. This is true for me for all versions since introduction.

What is very slow?

bubnikv, 2020-03-09:
We can reproduce this issue on mac of one of our college. We will debug this issue there.

kocikdav, 2020-03-10:
Hopefully 0836df9 fixes detection of SD cards on mac. It will be present in next RC. Im closing this issue. If the fix didnt help please reopen.

closed 2020-03-10 by kocikdav

aequitas, 2020-03-11:
Can confirm with rc3 it works as expected.

The-Mr-Fish, 2021-01-20:
I have had this same issue, and found that Google Backup and Sync was the cause of the issue and unticking "Plug in a camera or phone to back up files" in Backup and Sync fixes the issue. It even causes the issue when that drive is unticked in the list of backup targets.