prusa3d/PrusaSlicer #3795: 2.2.0-RC Eject is very slow in Linux

FlailAway, 2020-03-07:
### Version
2.2.0-rc

### Operating system type + version
Mint 19.3+Cinnamon

### 3D printer brand / version + firmware version (if known)
Anycubic Delta
Marlin 1.1.9
MKS-Gen-L

Love the new icons to Save-to-SD and Eject, but usually the first time after a new item slice, the Eject can take up to 12-seconds. For subsequent slicing on the same item, but changed print parameters it is only a second to two.

If it has to take that long the first time then a Progress Bar to let me know it is Ejecting would be handy so I do not try to click the icon again.

FlailAway, 2020-03-07:
It also does the long delay after using Add-instance for 2 more items for a total of 3 and slicing again.

bubnikv, 2020-03-08:
I am lost. Ejecting a slow external media may take time if the data is
still in the operating system cache. This is the way linux works. If the
operating system reports to slicer that the write is finished, it is
actually not. The data is still in the OS cache while it is slowly being
written. The eject operation then has to wait for the operating system to
flush the caches.

> It also does the long delay after using Add-instance for 2 more items for
a total of 3 and slicing again.

This may be the same. The operating system or the SD card itself may
optimize writing. If you save the same file twice, it may actually just
verify that the data is already there. This makes sense because writing is
usually slower than reading.

FlailAway, 2020-03-08:
@bubnikv Thanks for the reply, I had considered the buffering, but only PrusSlicer does this slow thing. If after a slice and Export GCode, I immediately go to the File Manager in another workspace with the SD card already in focus and eject, there it is instant. 

I hate to mention other slicers, but Cura does it in around a second like the File Manager. I am not interested in using Cura, I just tried it to see and it saves a 3MB file in about 3-seconds and ejects in 1-second.

It is clearly something PrusaSlicer is doing. It is not a big deal to me, but this being an "RC" version I figured you would want Comments.

FlailAway, 2020-03-08:
OK, some figures and 860kb .obj

In PrusaSlicer I click "Slice" then the SD/USB icon and wait until the Eject icon appears. I click the Eject icon+start the stop-watch and I get the "you can remove.." message at 17.3 seconds.

With PrusaSlicer again, but this time when the Eject icon appears I swap workspaces and click "Eject" in the File Manager (Nemo) 3.4 seconds.

Same process for Cura and I get "remove..." at 6.53 seconds.

It ain't normal with PrusaSlicer.

bubnikv, 2020-03-08:
PrusaSlicer enumerates all drives in your system every 2 seconds in a
background thread. It may be that you have something in your system that
responds slowly to enumeration.

We may look into Cura implementation for advice.

bubnikv, 2020-03-11:
We reworked the eject function to show a wait clock while waiting for the SD card / Flash Drive unmounting and to show the eject dialog immediately after the unmount command finishes. This behavior is very similar to what Cura does. I am afraid we cannot do more and based on our testing the unmounting is quick.

Closing. The fix will be in PrusaSlicer 2.2.0-rc3.

closed 2020-03-11 by bubnikv