prusa3d/PrusaSlicer #3797: Options don't always "take" on macOS

GarthSnyder, 2020-03-08:
### Version
PrusaSlicer-2.2.0-rc2

### Operating system type + version
macOS Catalina 10.15.2

### 3D printer brand / version + firmware version (if known)
Not relevant - issue is reproducible with PrusaSlicer alone

### Behavior
I've been noticing throughout the beta cycle that sometimes option settings don't seem to go into effect properly. But it's been hard to pin down exactly where and when, since often they work fine.

I've discovered through experimentation that the appearance of the "revert backloop arrow" icon in the settings UI is a pretty reliable indicator of whether the setting actually takes effect. If the icon appears, it means PS has acknowledged the change and the next-sliced G-code file will reflect the new value of the option. If this icon does not appear after a change, it means the change is still trapped at the UI layer and PS will ignore it.

The problem seems to randomly happen or not. It often goes in "runs" - if one option works, tabbing to the next and entering a new value usually works, too. And the converse is true as well; if one option doesn't take, the next one probably won't either.

As far as I can tell, the problem seems to apply to all option text boxes everywhere. For example, the Infill% text box in the main UI right column might or might not respond correctly to a change. I don't detect any particular pattern to the success or failure of individual options, but in aggregate, the issue is 100% reproducible for me.

I've uploaded a short video [here](https://youtu.be/NTIkB88Ykuw) that demonstrates the issue. Four items don't work, then the next few do, then the originals start working again too.

_Is this a new feature request?_
No

brettvitaz, 2020-03-09:
I have found this as well, especially when I am manually typing in values. I found the most reliable workaround is to make sure I hit enter every time I type a new value. Hitting TAB to select the next value does not make it work reliably. It seems to be intermittent.

It is obvious that it didn't work when I go back to the Plater tab and don't have to reslice.

GarthSnyder, 2020-03-09:
Hmm, interesting! Hitting Enter does seem to have a higher success rate for me, but it's still not 100%.

bubnikv, 2020-03-09:
@YuSanka suspects the following
commit: d1e343595680a40e4811914bdcef367b836c7f0e
The commit happened between alpha4 and the first beta. Would you guys
please test with PrusaSlicer 2.2.0-alpha4 whether the issue is there or
not? Thank you.

GarthSnyder, 2020-03-09:
Alpha4 works correctly for me.

bubnikv, 2020-03-10:
Thanks. We were able to reproduce the issue, @YuSanka is looking into it.

bubnikv, 2020-03-11:
This should be hopefully fixed in PrusaSlicer 2.2.0-rc3. Thanks for heads up.
Closing.

closed 2020-03-11 by bubnikv