prusa3d/PrusaSlicer #3798: Offset Print area origin doesn't match with custom bed model origin.

photodude, 2020-03-08:
### Version
Build:     PrusaSlicer-2.1.1+win64-201912101512
and
Build:     PrusaSlicer-2.2.0-rc2+win64-202003071138

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 34,307MB
OpenGL installation
GL version:   4.6.0 NVIDIA 441.87
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1070/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### 3D printer brand / version + firmware version (if known)
BIBO2 Stock model with 5015 part cooling blower and PEI sheet build surface
BIBO MFG provided stock Marlin 1.0.0 shaqFoo Ditto printing fork firmware

### Behavior
When using a specialized offset print area smaller than the printer bed model for ditto printing (copy/duplication printing with a dual extruder), the specified offset print area origin does not line up with bed model origin. The print area center lines up with the bed model center. 

* _Steps needed to reproduce the problem_

1. create a print area 33mm wide in X and 186mm in Y
2. set origin to right of print area (+33mm x)
3. create a bed model 214mm x and 214mm y
4. observe print area origin relative to bed model center

* _Expected Results_
specified offset Print area origin should line up with bed model origin, or should be able to be offset
* _Actual Results_
the current behavior is the print area center is lining up with bed model center.  There is no method other than setting the print area origin to specify where the print area should align relative to the bed model.
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
in the screenshot there is a small cylinder in the back showing the horizontal center. The center of the bed is the printer origin. The center of the offset print area can be seen in the center of the bed with the origin to the right of center. The print area should all be to the left of center as it's offset for ditto duplication printing
[image]

_Is this a new feature request?_
if needed, an option to offset the offset print area to match the origin of the bed model

Ditto printing printer configuration and model can be found in my Github Repo
https://github.com/photodude/BIBO2-PrusaSlicer

Related issues
#3560 for creation of vendor profile instructions

bubnikv, 2020-03-11:
Did you resolve this issue?
By the way, your profile is now in the master and we did some more final touches to it, you may want to update the wiki.

photodude, 2020-03-11:
@bubnikv I don't have a solution to this issue with the ditto printing restricted print areas and the bed model.

Awesome to hear that the profile is merged. I'll see if I can get to a few changes I mentioned in the PR https://github.com/photodude/BIBO2-PrusaSlicer/pull/4#issuecomment-597162948 sometime this afternoon. I'll see about updating the wiki with notes on printer variants in the method @rtyr  mentioned.

photodude, 2020-03-12:
@bubnikv My supposition on this issue is that it's related to an assumption in the bed model and bed texture rendering positions in this file https://github.com/prusa3d/PrusaSlicer/blob/master/src/slic3r/GUI/3DBed.cpp
GUI C++ is something new to me so It's going to take some time for me to parse what things mean to see if that's where the source of the issue is.

bubnikv, 2020-03-12:
@enricoturri1966 implemented alignment of the STL to the image by aligning center of one to the center of the other I believe.

photodude, 2020-03-14:
@bubnikv Yes, I think the alignment @enricoturri1966 implemented is the source of this. I'm leaning towards it being in this section of code https://github.com/prusa3d/PrusaSlicer/blob/master/src/slic3r/GUI/3DBed.cpp#L543-L567 probibly need some kind of condition for cases where the bed model and the print_area/bed_texture are not the same size. but it may be in the texture side of the code.

Initial thoughts on cases that need to be covered or could be covered.
1. bed model and texture are the same size
  - existing case
2. texture/print_area are smaller than the bed model with offset origin
  - this is the case with the ditto printing, the ditto printing area is either the left or right of bed model origin
3. texture/print_area are smaller than the bed model with offset origin
 - similar to 2 but the offset for the print area origin is far from the bed center (this is more of a thought of what could be set up, but I don't know of anything that does have origins this way)

I'll have to think on how to to address adding that second case

photodude, 2026-09-25:
dealing with duplication printing and duplication printing previews and how the bed should look based on the nozzle offset is still needed