prusa3d/PrusaSlicer #3799: Custom Printer SVG bed texture saved from AI fails to display correctly

photodude, 2020-03-08:
### Version
Build:     PrusaSlicer-2.1.1+win64-201912101512
and
Build:     PrusaSlicer-2.2.0-rc2+win64-202003071138

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 34,307MB
OpenGL installation
GL version:   4.6.0 NVIDIA 441.87
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1070/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### 3D printer brand / version + firmware version (if known)
BIBO2 Stock model with 5015 part cooling blower and PEI sheet build surface
BIBO MFG provided stock Marlin 1.0.0 ShaqFoo Ditto printing fork firmware

### Behavior
When using a custom bed texture saved from Adobe Illustrator the colors get inverted and lines disappear. The specific texture was created by modifying an existing SVG texture mk3.svg to remove the text to test custom SVG textures.

* _Steps needed to reproduce the problem_

1. in adobe illustrator modify an existing SVG texture mk3.svg to remove the text and make the size smaller for a 214mm by 186mm bed
2. save as SVG or export as SVG
3. load to a printer
4. observe bed texture is not white but black and the lines are missing.

* _Expected Results_
modified SVG should display correctly same as a PNG exported from the same file.
* _Actual Results_
the current behavior the color gets inverted and the lines disapear.
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
Without custom bed texture
[image]
With custom bed texture SVG that does not display correctly
[image]

_Is this a new feature request?_
documentation on any specialized settings for SVG textures to ensure they display correctly.

The printer configuration and SVG bed textures can be found in my Github Repo
https://github.com/photodude/BIBO2-PrusaSlicer

Related issues
#3560 for creation of vendor profile instructions
#3798 for the origin of custom offset print areas relative to bed model origin

mikolaszuza, 2020-03-09:
I've added a relevant article to the documentation:
[Custom SVG and PNG bed textutes](https://help.prusa3d.com/en/article/custom-svg-png-bed-textures_124612)

TLDR:
The Nano SVG parser, which PrusaSlicer uses, does not support every feature of the SVG format. We suggest the following workflow to correctly export SVG texture from Adobe Illustrator:

File - Export - Export As
Tick Use Artboards to make sure that the resulting size is correct
In the SVG Options dialog pick:
Styling - Inline style (Internal CSS is NOT supported)
Confirm with OK

Let us know if this fixes the problem for you, thanks!

photodude, 2020-03-09:
@mikolaszuza Awesome, that is the key information that I needed, with those changes to the SVG export everything is now working. I'll add that link to the vendor configuration bundle stuff in the wiki.

As my issue is solved and documentation for SVG export settings are provided, I'll consider this issue closed.

closed 2020-03-09 by photodude

mikolaszuza, 2020-03-09:
Great! Happy to hear that. And thanks for bringing the attention to SVG export.