prusa3d/PrusaSlicer #3805: purge extruders with origin center is off printing area for dual extruders

photodude, 2020-03-09:
### Version
Build:     PrusaSlicer-2.1.1+win64-201912101512
and
Build:     PrusaSlicer-2.2.0-rc2+win64-202003071138

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 34,307MB
OpenGL installation
GL version:   4.6.0 NVIDIA 441.87
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1070/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### 3D printer brand / version + firmware version (if known)
BIBO2 Stock model with 5015 part cooling blower and PEI sheet build surface
BIBO MFG provided stock Marlin 1.0.0 shaqFoo Ditto printing fork firmware

### Behavior
When the origin is in the center of the bed the purge all extruders results in the purge line being at 0 Y and 50% X length. unfortunately this results in the tool path being outside the print area.

* _Steps needed to reproduce the problem_

1. multi extruder print (in my case a 2 color print for dual extrusion)
2. set origin at center of the bed
3. turn on wipe tower and purge all printing extruders
4. slice
5. tool path outside the printable area.

* _Expected Results_
Purge line should be at the front of the bed and inside the printable area.
* _Actual Results_
the current behavior is the purge line is at 0Y and 50% bed length for X resulting in the purge line in the middle right of the bed with 50% of the purge line outside the printable area.

_Is this a new feature request?_
A re-positionable purge line, like the wipe tower can be repositioned, would be helpful.

printer configuration can be found in my Github Repo or the open PR #3804 
https://github.com/photodude/BIBO2-PrusaSlicer

**Project file**

[bulbasaur_dual_detail.zip](https://github.com/prusa3d/PrusaSlicer/files/4309264/bulbasaur_dual_detail.zip)

lukasmatena, 2020-03-10:
You are right, this was in there for quite a long time (probably always). The priming lines were printed with respect to coordinate system origin, not to "bottom-left" corner of the bed.

I hope I fixed it with a48a796. The fix will be in 2.3 release - 2.2 is now too close.

Closing the issue.

closed 2020-03-10 by lukasmatena

photodude, 2020-03-10:
@lukasmatena Awesome, thank you for fixing that. Sorry, I wasn't able to fully write up the Issue I originally mentioned in [this comment](https://github.com/prusa3d/PrusaSlicer/issues/3560#issuecomment-587537739) until now. Too bad it won't be in 2.2.0 but at least it will be fixed in the future.

lukasmatena, 2020-03-11:
After giving it a bit more thought, we decided to squeeze the change into 2.2.0. It is now in master as a6a7736. Hopefully it won't break anything.

photodude, 2020-03-13:
@lukasmatena Awesome I'll see about giving 2.2.0-rc3 a try once I do some maintenance on my printer. (time to replace some fans)