prusa3d/PrusaSlicer #3806: wipe tower interferes with print object when arrange all objects is used

photodude, 2020-03-09:
### Version
Build:     PrusaSlicer-2.1.1+win64-201912101512
and
Build:     PrusaSlicer-2.2.0-rc2+win64-202003071138

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 34,307MB
OpenGL installation
GL version:   4.6.0 NVIDIA 441.87
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1070/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### 3D printer brand / version + firmware version (if known)
BIBO2 Stock model with 5015 part cooling blower and PEI sheet build surface
BIBO MFG provided stock Marlin 1.0.0 shaqFoo Ditto printing fork firmware

### Behavior
When a multi-extruder object has a wipe tower and the "arrange all objects is used the wipe tower can end up interfering (overlapping) the print object.

* _Steps needed to reproduce the problem_

1. multi extruder print (in my case a 2 color print for dual extrusion)
2. turn on wipe tower
4. press the "arrange all objects
5. observe wipe tower interfering with the object 

* _Expected Results_
Wipe tower should be positioned at a location near but not overlapping the print object.
* _Actual Results_
the wipe tower often overlaps the object
* Screenshot
[image]

printer configuration can be found in my Github Repo or the open PR #3804 
https://github.com/photodude/BIBO2-PrusaSlicer

**Project file**
wipe tower before using arrange all object
[bulbasaur_dual_detail.zip](https://github.com/prusa3d/PrusaSlicer/files/4309264/bulbasaur_dual_detail.zip)
wipe tower after arrange all objects 
[bulbasaur_dual_detail_wipeTower_interferance.zip](https://github.com/prusa3d/PrusaSlicer/files/4309302/bulbasaur_dual_detail_wipeTower_interferance.zip)

lukasmatena, 2020-03-10:
This should be already fixed in 2.2.0-rc2.

lukasmatena, 2020-03-10:
Actually, the arrange function accounts for the wipe tower since 2.0.0, so it's not something that was fixed recently. Something is clearly wrong, but I have no clue what it is. I didn't manage to reproduce the issue (at least not on Linux). We will try to reproduce and fix it.

peymon, 2020-03-20:
I can confirm, I have the same issue. Had it since 2.2.0 RC3. It has not been fixed in RC5.

tamasmeszaros, 2020-03-20:
Hi @peymon ! Please, could you send a 3mf where the issue occurs? I could not reproduce with @photodude 's  project files.

peymon, 2020-03-20:
Hi @tamasmeszaros,

The issue occurs regardless of the model. I just reproduced it using the Prusa Beer Opener and attached it for you here. Please note that I am using the Mac version: 2.2.0-rc5+-202003191653. But I have had this issue since rc3. I have not tested it on previous builds.

[image]

[PrusaBeerOpener.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4360674/PrusaBeerOpener.3mf.zip)

bubnikv, 2020-03-20:
I am not able to reproduce this issue with PrusaSlicer 2.2.0-rc5 either. To be sure, I tried on both Windows 10 and OSX Catalina.

peymon, 2020-03-20:
I tried to make a gif of the problem. See attached, please.

[image]

photodude, 2020-03-20:
@bubnikv I'm guessing it requires having the wipe tower selected when arrange is selected. 
Which of the 2 project files I uploaded did you try?

bubnikv, 2020-03-21:
I tried both, I tried even selecting different parts, I even tried to run
the partial arrangement (Shift-A), but I was not able to reproduce this
issue.

I will once again with the wipe tower selected.

bubnikv, 2020-03-21:
I was not able to reproduce the issue with PrusaSlicer 2.2.0-rc5 with any models provided. However these are my hints:

The wipe tower has two states:
1) G-code not generated yet, thus the size of the wipe tower is not know yet and the wipe tower block shows the two zigs.
2) G-code generated, the wipe tower size is known and the rear wall of the wipe tower is rendered flat without the two zigs.

The animated gif by https://github.com/prusa3d/PrusaSlicer/issues/3806#issuecomment-601838900
shows a wipe size was known. I was thinking maybe the issue could be that the arrangement will invalidate the wipe tower from state 2) to 1), but it does not, as just moving the objects around does not change the amount of wipe (this assumption may be no more valid after PrusaSlicer 2.2.0 path planning change though).

The wipe tower position is stored in the Print preset. This is not optimal, it is a heritage of the Slic3r PE impelmentation we did before we had 3MFs. I would expect that the wipe tower could jump back to the position stored in the Print preset if some messages indicating the wipe tower position inside PrusaSlicer were delivered in a wrong order. But this does not seem the case here either.

bubnikv, 2020-03-22:
By the way, do you guys have background processing enabled?

photodude, 2020-03-22:
@bubnikv I do not have background processing enabled.
[image]

photodude, 2020-03-22:
@bubnikv This is what I got with a dual extrusion calibration target, background processing, and wipe tower using auto arrange. Clicked on bed before auto arrange to deselect everything.  2.2.0-rc5 BIBO dual printer, 0.15 print profile
[image]

before
[wipe_tower_error_before_Dual_Extruder_Calibration_Target.zip](https://github.com/prusa3d/PrusaSlicer/files/4365276/wipe_tower_error_before_Dual_Extruder_Calibration_Target.zip)

after arrange
[wipe_tower_error_after_Dual_Extruder_Calibration_Target.zip](https://github.com/prusa3d/PrusaSlicer/files/4365277/wipe_tower_error_after_Dual_Extruder_Calibration_Target.zip)

photodude, 2020-03-22:
@bubnikv I have discovered the replication step that was missing, =D
happens only when "auto center parts" is enabled.

Seems there is a missing, or incorrect, collision check for the wipe tower when "auto center parts" is enabled.

with the before project uploaded above and "auto center parts" is enabled there is an issue (see image in post above), but disabling "auto center parts" keeps the wipe tower positioned without collision with the model.
[image]
Re-enabling "auto center parts" causes the issue again.

bubnikv, 2020-03-22:
Cool. The auto-centering feature is posed for removal in the next version.
We will rather auto-arrange the newly added objects to the print bed
automatically while keeping the objects already on the print bed static.

peymon, 2020-03-22:
@photodude, genius! Indeed without auto-center, the issue is no longer there.

photodude, 2020-03-29:
@bubnikv When "auto-center" is removed will it be replaced with an "auto-arrange"?
It would be nice for models to auto-arrange when first placed on the print bed, then let me adjust positions as I see fit.

photodude, 2020-03-29:
@bubnikv ::face-palm:: ... I see you already said that auto-arrange on first add is coming and I forgot to reread before commenting... sorry.

photodude, 2026-09-25:
close as stale. auto-centering feature was removed as noted above and does not apply to newer versions.