prusa3d/PrusaSlicer #3807: Unify "MK3S" and "MK3S MMU2S Single" profiles thereby removing any distinction between printer types

nordurljosahvida, 2020-03-10:
### Version

2.2.0-rc2

### Operating system type + version

Ubuntu 18

### 3D printer brand / version + firmware version (if known)

MK3S MMU2S

### Behavior

* _Describe the problem_

PrusaSlicer distinguishes between `MK3S` profiles [for a standard, non-MMupgraded printer] and `MK3S MMU2S Single` for single-color prints on a MMU printer. These profiles are - according to customer support on shop.prusa3d.com - exactly the same profile, except for:

1. The filament purge at the beginning of the print [`MK3S` does a single, half-width line whereas `MK3S MMU2S Single` goes fullwidth and back, therefore tracing two parallel and adjacent lines]

2. `MK3S MMU2S Single` allows you to choose which of the 5 filaments in your MMU printer you'd like to execute the print with, whereas launching a print sliced for `MK3S` on a MMU printer will A] warn you about the `wrong printer type` and *not* allow you to choose the filament, and it'll [sometimes] use the filament that's already loaded to nozzle.

As far as I can tell, there is absolutely no reason to distinguish between these. At our print shop, maintaining single-color prints sliced for our `MK3S MMU2S` printers separate from our `MK3S` prints is a mess, even though they are **the exact same types of print**. And for any user on earth, I think, it would be so much simpler to have a unified single-color print profile, that executes perfectly on standard, non-MMU machines [MK3S] and also on MK3SMMU2S machines. 

* _Steps needed to reproduce the problem_

Use Prusaslicer with a MMU prusa printer

* _Expected Results_

PrusaSlicer offering a single, simple, single-color MK3S profile that also works on MMU2S machines, that will ask you which of the 5 filaments to use, *not* warn about incompatibility [as they are the same printer with just an upgrade that does not affect the printing profile in any way shape or form] and proceed as normal with a single-color print.

This could be achieved by:

1. eliminating `single` profiles in MMU printers. Adding a MMU printer would only add truly MMU profiles.
2. making MK3S profiles show as `MK3S / MMU2S single material` with a simple aesthetic tweak when a MMU printer is selected, without actually using any different profile at all.
3. upgrading prusa-firmware to read MK3S prints on MK3SMMU2S printers as compatible [not warn about wrong printer model], prompt to select filament, and proceed with print as normal.

This way, all single-material prusa gcodes on earth would be exactly the same independently of whether they're sliced for a MMU printer or not.

To cite one example, this would really help simplify things for the teachers in the school where my store has placed a MK3SMMU2S printer for all their staff and students to use.

_Is this a new feature request?_

Yes

rtyr, 2020-03-11:
From the technical point of view, the differences between MK3S and MK3S MMU2S Single are:

**Start gcode**
_different intro line which is purging much more material compared to regular MK3_ - it is needed for printers with MMU upgrade, because there is different load mechanism (no purging and asking user if filament is clean like on regular MK3). Filament is loaded to the nozzle after mesh bed leveling, so it must be cleaned this way.

_firmware version_ - can be different from regular MK3S (not the case atm), but in general we might need different FW version announcement for different printer model or variant.printer model check - this check was introduced globally. It is possible, that we will make it less strict in the future. The main purpose was to distinquish between MMU2 and MMU2S, which are not compatible. But it proved to be quite usefull in general (for example if someone used downloaded 3MF with different printer profile selected).

_MMU specific gcode commands_ - Tx for filament selection and initial load, Tc for loading the filament to the nozzle (after MBL). These commands are not compatible with regular MK3S. I am not sure if it can cause any issues (with Octoprint etc.).

**End gcode**
_Custom ramming sequence for MMU_
_MMU specific gcode for unload filament to the MMU unit_

We prefer separate printer profiles that match our product line. There are several reasons for it and not all of them are technical. We want clear and simple workflow for new customers. That is one of the reasons why we have MK3 and MK3S separately even if the profiles are almost identical. It is better for our customer support as well.

Current state might not be perfectly ideal for everyone, but I don't see global need for changing it at the moment (considering it would need changes to PrusaSlicer, firmware and profiles). As far as I know, you are the first to mention this as an issue.

What we might consider is to make regular MK3S/2.5S printer profile more "MMU2 friendly" (add Tx and Tc commands which would be ignored if no MMU unit connected) and to make printer model checks less strict. Both things are mainly FW related.

Gcodes sliced for regular MK3 can be printed on MMU2 printer quite easily using "Load to nozzle" option on the printer before the print. Printer model check can be disabled on the printer side.

bubnikv, 2020-03-15:
@nordurljosahvida Is this explanation understandable? Could we close the issue?

nordurljosahvida, 2020-03-15:
@rtyr I see, and I do think making changes fw-side to make this as transparent as possible regardless of the general choice would be ideal. Asking for filament on a MMU2S printer when printing a MK3S gcode if none of the 5 filaments is loaded should be implemented i think. And why not leaving it up to the printer to understand how much filament to purge and when [before or after the MBL]?

In general, as much as I understand your desire to keep your product line distinct, why not try to make the produced gcodes the same regardless behind the scenes?

During PS 2.2.0 development there have been numerous changes regarding the logic of "this is a MMU printer in single mode, so shall we offer the user the opportunity to change colors at a certain height by changing extruder anyway?" and "this feature previously available only for MK3S is now available for MMU2S single mode but these others have been dropped and are only available in MMU2S multi mode".

Won't it become increasingly harder to track software logics in PS in the future if we keep everything structurally separate rather then a surface, aesthetic difference where the user simply selects their printer but identical gcodes are produced behind the scenes? Wouldn't this simplify things for CS as well?

@bubnikv absolutely, feel free to close this, i don't want to take time away from other things.

nordurljosahvida, 2020-03-24:
@rtyr @bubnikv It just occured to me that the single largest usecase for this FR is prusaprinters.org. I am now downloading the [3d printed visors prusa made for covid](https://www.prusaprinters.org/prints/25857-prusa-protective-face-shield-rc2/files) and i'm having to reslice them [with two instances of PS open, one for reference, the other with MMU2S printer selected since changing printers resets any customizations to the print settings, therefore requiring me to apply them by hand one by one] rather than simply printing the already included gcode, even though we have the exact same printer! Think of all of the MK3S prints out there that can't be easily printed on MMU2S [because the lack of filament on print start is very poorly handled for example] and viceversa! I really think we should make gcodes exactly the same behind the scenes so that they're interchangable.

bubnikv, 2020-03-24:
I feel and share your frustration, but as stated by @rtyr we need a
firmware support to unify these profiles in PrusaSlicer.

You may express your requests to the firmware team here:
https://github.com/prusa3d/Prusa-Firmware/issues

Likely there may already be a similar issue open, so please search there
first.

rtyr, 2020-03-25:
> 
> since changing printers resets any customizations to the print settings

That is not correct, you can switch between MK3S and MK3S MMU2S Single without losing any print settings customizations. Try it with PrusaSlicer 2.2.0 final.

nordurljosahvida, 2020-03-29:
> I feel and share your frustration, but as stated by @rtyr we need a firmware support to unify these profiles in PrusaSlicer. You may express your requests to the firmware team here: [...]

thanks, still looking through their issues to ensure i don't open a duplicate.

> > since changing printers resets any customizations to the print settings
> 
> That is not correct, you can switch between MK3S and MK3S MMU2S Single without losing any print settings customizations. Try it with PrusaSlicer 2.2.0 final.

True, is this a new feature?

Thanks

rtyr, 2021-08-12:
I believe we can close this. I don't see an implementation of something like this realistic. It would require inadequate amount of effort on several fronts and I don't think it's actually needed.

Closing.

closed 2021-08-12 by rtyr

nordurljosahvida, 2021-08-13:
@rtyr this also closes #3927, we don't want to explore that even, right?