prusa3d/PrusaSlicer #3808: Convert 100% internal infill to solid infill

espr14, 2020-03-10:
v2.2.0-rc2 win64

Solid infill and 100% internal infill has the same pattern and infill density but they are kept separate. This prolongs print time and G code while making the print weaker. 

100% internal infill should be converted to solid infill during slicing. If somebody wants to keep them separate they can choose 99% infill density which is nearly the same.

[image]

bubnikv, 2020-03-10:
the two infills on your screenshots are very different.

espr14, 2020-03-10:
Different extrusion width.

Wider extrusion brings:
- faster print
- stronger print
- faster recovery of bent ceilings

bubnikv, 2020-03-11:
> Different extrusion width.

An incomplete sentence is not informative. I have no idea what you want to say with this incomplete sentence.

> Wider extrusion brings:
> 
> faster print
> stronger print
> faster recovery of bent ceilings

So what? As usual, I don't understand what you are proposing.

> Solid infill and 100% internal infill has the same pattern and infill density but they are kept separate. This prolongs print time and G code while making the print weaker.

But on your picture you show extrusions with very differing extrusion widths? I cannot make sense of it.

> 100% internal infill should be converted to solid infill during slicing. If somebody wants to keep them separate they can choose 99% infill density which is nearly the same.

I can agree with that, but this is not what your picture shows.

espr14, 2020-03-11:
I propose to merge infills. Image is illustrative.

bubnikv, 2020-03-11:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/107

closed 2020-03-11 by bubnikv