prusa3d/PrusaSlicer #3809: 2.2.0-rc2+ Issue / Question with Brim printing direction

tilllt, 2020-03-10:
<details><summary>Sysinfo</summary>
<p>

PrusaSlicer
Version:   2.2.0-rc2+
Build:     PrusaSlicer-2.2.0-rc2+-202003071126

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Sierra Version 10.12.6 (Build 16G2016)
Total RAM size [MB]: 8,590MB
OpenGL installation
GL version:   2.1 NVIDIA-10.18.5 378.05.05.25f16
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce 9500 GT OpenGL Engine
GLSL version: 1.20
Installed extensions:

GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_depth_bounds_test
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_fragment_program2
GL_NV_fragment_program_option
GL_NV_light_max_exponent
GL_NV_multisample_filter_hint
GL_NV_point_sprite
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod

</p>
</details>

Prusaslicer is changing print direction in the middle of printing a brim. This leads to "lifting" the already printed brim on the junction point of the two lines and therefor reduced adhesion of the Brim. What is the reason for not printing the entire brim with the same printing direction and can it be turned off or is this intended behaviour?

JohnnyDeer, 2020-03-11:
Hi, skirt is placed in brim, that´s reason why print orientation change. Try to set skirt more distanced from object, to not be inside of brim, that should solve your problem.

tilllt, 2020-03-11:
The only thing i changed from the default settings is that i enabled the Brim.

tilllt, 2020-03-11:
So, while your suggestion may help me with my problem, the question would be if the default settings should be changed.

JohnnyDeer, 2020-03-11:
In some situations is wanted to have brim around skirt. For example if you print big object from ABS... you set heigh skirt, for better wind resistance and skirt should be firmly on bed. Then you want brim around skirt. So IMHO default settings should not be changed.

tilllt, 2020-03-11:
I think if it's desired in "some" cases, but usually not, it should be enabled for those cases and not needing to be disabled for 90% of the other cases. How many people print abs with a draft shield?

tilllt, 2020-03-11:
Let's out it this way, the standard user expectation is to have either skirt or brim, not both.

JohnnyDeer, 2020-03-11:
In this case user should disable skirt and enable brim. We can include this information in slicer manual, to avoid situations like yours. 
If you would like us to change default settings, than please change this issue to feature request.

tilllt, 2020-03-11:
Look, i am giving you feedback that this is unexpected behaviour for the average user (like me) and a lot of people will run into the same error. A LOT of people use brims, since its a sureshot way to prevent warping and help adhesion. Thats just my experience from my personal learning curve and from discussing with others. So, i am giving you the feedback: The way that the new version of Prusaslicer handles a VERY common setting (aka enabling brim) is leading to unexpected results.

Thats it. You can use this information to help others to NOT run into the same problem or you dont. Adding it to the documentation will definitely not help at all, i guess you know how many people RTFM.

JohnnyDeer, 2020-03-11:
I understand you, we are happy for all feedbacks, thanks to it we can go forward. I tried to answer your question, why is brim changing direction of print and give you solution. 
Changes in brim and skirt was added in version 2.2.0 alpha1, based on FR #3634 , thats why behavior is different from older versions. Changing of print direction is known to us, we can have discusion abou if it should be changed. So thats why you can change this issue to FR.

rtyr, 2021-02-08:
The current behavior is already discussed here https://github.com/prusa3d/PrusaSlicer/issues/3692. We are investigating possible ways to improve it. Closing as a duplicate.

closed 2021-02-08 by rtyr