prusa3d/PrusaSlicer #3810: [feature request]

marcottt, 2020-03-10:
### Version
any

### Operating system type + version
any

### 3D printer brand / version + firmware version (if known)
any

### Behavior
When brim start, sometimes nozzle is dirty from previous plastic or blob or similar, these blob go to bed and every round of brim dirty again the nozzle and then the print object.
I suggest an "helical brim"  so external round of brim is distant (10/15/30 mm ?) each other to have nozzle to stay clean....  

_Is this a new feature request? YES

#### Project File (.3MF) where problem occurs

JohnnyDeer, 2020-03-11:
Hi, thank you for your suggestion. You can kinda do your idea with skirt. If you will place skirt in sufficient distance from object and set height to 1 layer, dirt from nozzle can be catched in skirt.

marcottt, 2020-03-12:
i already do it... but some times blob are get from one skirt line to the adiacent and go on... because line are adiacent... and is useless they stay adiacent... i think maybe better have 5/10mm (for example) distance between each skirt line.

closed 2026-09-29 by prusaslicer-bot[bot]