prusa3d/PrusaSlicer #3811: [Featue request] Make partial Notes visible on main page

FlailAway, 2020-03-10:
### Version
2.2 and beyond

### Operating system type + version
Any

I really like the "Notes" section in each of the three departments, Print, Filament and Printer. Please keep those 3 disparate sections as I rarely change the Filament and Printer when compared to the Print" options.

What would be really handy would be the have a **single** Text-line under each of those 3-items on the Home page and display part of the first line of the Note.

May Saving naming system is 
Print Settings = "_0.2,3+3+3,@100,15%_" = nozzle, walls, top, bottom,speed,infill
Filament = "_tpu,215=230+55_" = plastic,hot-end temp,bed temp (thermistor is off 15-deg)
Printer = "_anycubic-delta-01_" = using this printer

As in,
Print Settings:
0.2,3+3+3,@100,15%
Cubic infill with Brim

Filament:
tpu,215+55
The thermistor reads 15-degrees low, so set temp down 15-deg from required temp

Printer:
anycubic-delta-01
This is the all-metal modified delta

The thermistor Note would be handy to see for the filament instead of including the 215=230 in the saving name.

bubnikv, 2020-03-11:
This is very much a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1697

FlailAway, 2020-03-11:
Not even close as the information is always visible without having to wave the mouse over it. Also tool-tips do not work well with a touch-screen.

Not sure why I bother mentioning this stuff to help improve the slicer if it is just going to sloughed off defending the indefensible.

closed 2020-03-11 by FlailAway