prusa3d/PrusaSlicer #3812: Remove Logo when placing supports

andrewluebke, 2020-03-10:
### Version
2.2.0-rc2

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa SL1

### Behavior
It would be nice if the logo was hidden when placing supports.  The grid is still nice for lining things up but the logo gets in the way of following edges.

[image]

_Is this a new feature request?_
YES

foreachthing, 2020-03-10:
Create your own logo or remove it manually:
[image]

andrewluebke, 2020-03-10:
Good thinking, thanks.  I'll close this...

closed 2020-03-10 by andrewluebke