prusa3d/PrusaSlicer #3813: Possible issue induced by slicer (extreme vibration and blobs)

italocjs, 2020-03-10:
### Version
2.2.0 RC2

### Operating system type + version
Windows 10 1909 

### 3D printer brand / version + firmware version (if known)
3D Tecnologia Office, using marlin 2.0.x bugfix from around january 15, 2020 

### Behavior
Using exactly the same settings in two slices, produces different results, on prusa slicer the printed part appears to have extreme vibration and a lot of oozing, i tried my best to make a "fair" test, using the same material, same printer and configurations, but the results are way too weird.  i'm not sure what i could use to improve this  (i'm attaching my config bundle and stl if someone wants to try, the idea of this part is to measure XY and verify the surface quality.

Left is from Simplify3D,  right is from Prusa Slicer 2.2.0 RC2

[image]

[Dimensional calibration (bad on pslicer).zip](https://github.com/prusa3d/PrusaSlicer/files/4314065/Dimensional.calibration.bad.on.pslicer.zip)

One thing i was suspicious was about accelerations (which prusa slicer can control and S3D cant,  so i set both of them to the same acceleration before printing)

italocjs, 2020-03-11:
Well, i have "kinda" traced it down to something related to thin wall in prusa slicer, as the same model but with the wall a little bit bigger (was 0.48, its 0.6 now)  has printed without all the weird vibrations on pslicer.  I have tested the same model (0.6mm wall) on simplify3d and it also printed well.

Anyone has any idea on where i could improve this? i tried messing with mesh resolution but anything but the original settings break my model thin walls.

ManuelMcLure, 2020-03-11:
Try setting Gap Fill speed to 0 so there's no gap fill and see if that helps. I think gap fill tends to overextrude so you end up with a nasty outer surface.

rtyr, 2021-02-08:
Similar to #149.

italocjs, 2021-02-24:
Closing this,  i didnt noticed this in quite a white, probably already improved/fixed

closed 2021-02-24 by italocjs