prusa3d/PrusaSlicer #3814: [Feature Request] Do not create Bridge Infill on top of normal Infill

VFjr, 2020-03-10:
When there is a large flat surface, PrusaSlicer generates a Bridge Infill layer. I have my bridge fan speed set to 100% to help with actual bridges, but as a result I always get 100% cooling on the first layer on top of infill. 

This results on that layer not actually fusing together properly to the infill, and the overall strength of the part is reduced. 

My only option currently is to reduce the bridging fan speed, but in that case my actual bridges will suffer.

VFjr, 2020-06-11:
@jschuh Yes, that indeed addresses the issue and seems like a great implementation.

jschuh, 2020-06-12:
@VFjr, thanks for confirming.

@bubnikv, would it be possible to have someone take a look at this patch and see about getting it landed? I'm happy to make any additional changes if there are issues with how I implemented it.

closed 2026-09-29 by prusaslicer-bot[bot]