prusa3d/PrusaSlicer #3816: 2.2.0-RC2 - Panning/Moving the view leaves bed at weird angles

jweaver100, 2020-03-10:
### Version
2.2.0-RC2+Win64

### Operating system type + version
Win7

I am trying out 2.2.0 and my first reaction is that something has changed with the mouse/view behavour. Where as before the view seemed locked to the center of the bed so it always rotated around a fixed point.. now in 2.2.0 after moving the view around the bed ends off skewed in all directions and its really fiddly and frustrating to get it level again, which you do by randomly clicking and dragging the mouse until you get it correct.

I assume this has something to do with the 3DConnexion support, which is fine as I am a 3D mouse user. And I have to say the new functionality is nice when you use the 3D mouse feature.. But for a normal mouse its horrible.

I notice that there is a new "use free camera" option and wonder if this would fix what i am seeing, but the option does not appear to do anything.

To be honest, this is such an obvious change, I am sure its already been reported, but I just wanted to report this just incase no-one else has.

bubnikv, 2020-03-11:
When you have your 3DConnexion mouse plugged in, you may rotate the
viewport freely into configurations, which are not compatible with the
"constrained" camera rotation mode that you were used to before. We decided
to switch to the "free" camera rotation for the normal mouse in case you
have your 3Dconnexion mouse plugged in, so that the two devices have the
same degrees of freedom. Otherwise the viewport would have to jump abruptly
into the "Z up" orientation after your oriented the viewport with your 3D
mouse and then you touched the normal mouse and tried to adjust the
viewport with it.

You see there is a reasoning behind our current implementation. The
decision has been done between two bad choices. I personally have no strong
opinion which one is better.

jweaver100, 2020-03-11:
So this is a side effect of me having my 3D Mouse connected?

Is there really no solution other than disconnecting the mouse (which for me is near impossible)?

I will still use a mouse 95% of the time, but now this new feature has made the GUI unusable... I would rather turn off 3DMouse support all together and just go back to standard mouse.

This is going to be a major problem for people like me who have a 3D mouse but rarely use it... As now Slicer is going to be a nightmare to drive.

neophyl, 2020-03-11:
I have the same setup with a 3d mouse and when the view gets too skewed I just use one of the number keys like 0 or 1 to put it back to a known fixed angle and then manipulate again from there.

Just a workflow workaround.

jweaver100, 2020-03-11:
Thats a decent workaround. I will try...  Still not sure I like it as it will mean totally re-learning how to navigate the bed..

I was so looking forward to 3D Mouse support, but now I think I regret it as the functionality I use every day will change...

bubnikv, 2020-03-11:
You guys surprise me. You have the 3D mouse connected, but you rarely use
it. I hate the free camera rotation with the mouse, but then I expected
that somebody fluent with the 3D mouse, which allows one to skew the bed so
obnoxiously, is fluent with the free camera rotation with mouse, which is
BTW he default mode in Fusion 360. Now I am perplexed. What should we do to
make you happy?

jweaver100, 2020-03-11:
Don't base my comments as typical for 3D mouse users.. I like mine, but only use it for certain applications.. Its certainly better than mouse for viewing... But you can't do much else with it (i.e select options, or move objects) so for me, the mouse is still my preferred way of doing things.

I just use the 3D Mouse, when looking around the object..

So for me, I would prefer to keep the Mouse support them same.. And have 3D Mouse support as an option... But I can't speak for everyone.

Perhaps just having a reset/center view button would be a compromise... Not on the keyboard (1 or 0) but on the GUI...

neophyl, 2020-03-11:
I'm a new 3d mouse user so I haven't got fully used to it yet.  Still learning its ins and outs but I'm using it more and more in Blender.  
Unfortunately I only have it at home and as I swap between the work pc/home its not become second nature as yet.  So sometimes even at home I go for the mouse to pan the view around.  That's why I was aware of the issue but it doesn't bother me personally as I use my workaround and it just reminds me to use the input device I paid a pretty penny for :)

jweaver100, 2020-03-11:
you make a good point.. This issue will certainly force me to use mine more often.. But mine is an unsupported old type (SpacePilot) which has its share of problems.

And to be honest, whilst the 3D mouse is great for moving around the bed and inspecting the object.. its not great for placing items, moving them, and driving the GUI... So there are going to be times where a mouse is still the go to method of input...

bubnikv, 2020-03-12:
> it just reminds me to use the input device I paid a pretty penny for :)

And it is quite heavy indeed to the extent one can self defend with it.

enricoturri1966, 2020-03-13:
I made a build where the camera automatically switches back to constrained mode (if selected in the preference dialog) as soon as the user starts navigating (rotating/panning) the 3D scene using the regular mouse even if a 3DConnexion device is plugged in.
Could you guys please test it and tell me if you consider this an improvement or a regression ?

https://www.dropbox.com/s/o5srbn9c7i4db1l/PrusaSlicer-2.2.0-rc3%2B5-et-free-camera-linux-x64-asan-g4c16ce4-202003120927.AppImage?dl=0
https://www.dropbox.com/s/38vm4w4fgs584p0/PrusaSlicer-2.2.0-rc3%2B5-et-free-camera-asan-g4c16ce442-202003120915.dmg?dl=0
https://www.dropbox.com/s/qa920hn5r4k3de3/PrusaSlicer-2.2.0-rc3%2B5-et-free-camera-win32-g4c16ce442-202003120933.zip?dl=0
https://www.dropbox.com/s/hnqg1r41tnz7950/PrusaSlicer-2.2.0-rc3%2B5-et-free-camera-win64-g4c16ce442-202003121016.zip?dl=0

neophyl, 2020-03-13:
Will try it on my pc at home after work with the 3d mouse.  Just tried it at work with just a normal mouse just to get a baseline and operation looks normal (windows 10 64bit).  Rotate/Pan/Zoom ok.

jweaver100, 2020-03-13:
Just tested.. Its PERFECT... Its a tiny bit jarring when the pad snaps back to level when you touch the mouse, but its so insignificant its not even worth considering.

Thanks so much for doing this.. teh GUI is so much more usable now.

Just wait to get others opinions too, but I can't see anyone is going to be unhappy with this.

neophyl, 2020-03-13:
Tried this now at home and I agree with jweaver.

bubnikv, 2020-03-14:
great. Merged to master. It will be in the next rc.
Closing.

closed 2020-03-14 by bubnikv