prusa3d/PrusaSlicer #3817: Feature Request: Cura's "Make Overhang Printable", but to remove not add volume

haleypearse, 2020-03-10:
Cura's "Make Overhang Printable" adds volume under overhangs, apparently by projecting downwards from the top of overhangs at the "Maximum Model Angle." But what would be more useful is the opposite, to project up from the bottom of overhangs, removing volume from the object. 

Often for structural prints shaving off the overhangs would be far better than having droop. For example, this would automatically solve the issue of horizontal bolt holes sagging at the top, which necessitate time consuming model changes or drilling out. 

Ideally though, both options would be available, as both are bound to be useful.

This will droop and need drilling.
[image]

Here's what Cura does, which isn't useful here, but removing volume would make this globoid worm gear printable.
[image]

Ark-kun, 2020-03-11:
Interesting.
I was working on a different problem - better detection of "printing in mid-air" and the worst overhands.
While thinking of that I had an idea on how to use my proposed solution to fix the issue you describe. If a section of a tool path sticks out too much, just cut out the extra part. Although that might be harder in practice.

neophyl, 2020-03-11:
Interesting.  When designing specifically for 3d printing, especially when you know which way up you are going to orientate your part designing holes with a sloped 'roof' like top, as in your second example picture is fairly common practice.  
Its not very common on general designs or those downloaded from the usual internet sites though.  So some form of tool that would allow users to mark holes where this is required might be useful.

haleypearse, 2020-03-11:
> Interesting.
> I was working on a different problem - better detection of "printing in mid-air" and the worst overhands.
> While thinking of that I had an idea on how to use my proposed solution to fix [...]

Ark-kun, your solution sounds more generally useful. For instance probably you don't want to cut away bridges. But the algorithm "If a section of a tool path sticks out too much, just cut out the extra part" seems like a good and simple way to achieve this feature request. neophyl suggests the user is able to mark places, and I suppose the feature could be turned on with modifier meshes.

I'm an aspiring coder, but it isn't clear to me where to start with the source code.

Ark-kun, 2020-04-15:
I've created a PoC overhand detector in Python. It basically reads the gcode and rasterizes the layers. When rasterizing each line in the next layer it checks the support underneath the end point (so, the bridge is fine). If the support is lacking, the tool paints that part in a different color to indicate problem.

I think and algorithm like this would be a better way to detect the points that need support than the standard "slope detection" algorithm since the correct slope really depends on layer thickness and nozzle diameter.

Integrating something like this with Prusa Slicer's gcode.cpp is going to be painful though.

haleypearse, 2020-04-25:
Interesting. Couldn't the layer thickness and nozzle diameter also be parameters of the slope detection algorithm? I don't understand why rasterizing should be helpful at any stage since slicing converts points in space (stl) to points in space (tool paths).

haleypearse, 2020-11-23:
A simple way to accomplish this would be this: 

Whenever a perimeter would appear as an overhang perimeter (blue), recede it until it doesn't overhang. On each layer the outer perimeter will try to fill the area dictated by the model at that height, unless it would overhang. Eventually, moving up the layers, the perimeter will catch up to the model. Bridges would have to be detected specifically as an exception to this.

haarp, 2022-07-14:
This is #2575

stepikovo, 2022-07-18:
I'm closing this as a mentioned duplicate

Ark-kun, 2022-08-13:
>I don't understand why rasterizing should be helpful at any stage since slicing converts points in space (stl) to points in space (tool paths).

These points are not really points. First of all they are tool paths. - 1D objects, not 0D.
These "points" are flat cylinders of filament. They have radius. On layer 1 you have a point with coordinated {1,1,1}. On layer 2 you have a point at {x=1,y=1.2,z=1.4}. Whether the 2nd point "lays" on top of the 1st point depends on the nozzle size. The easiest way to check the overlap would be to rasterize these points as circles with radius=nozzle_diameter. One point/cylinder can be supported by multiple cylinders/paths under them. The best way to check the overlap is to rasterize (say, using a nozzle_diameter/10 grid).