prusa3d/PrusaSlicer #3818: New Feature Request - Allow custom filament settings to be used on different printers

DRKMSTR, 2020-03-11:
Currently when I set up a new filament (Let's say, "eSUN PLA+"), it only allows me to use it with the printer I set it up with. I would like to use it across all my printers.  

I'm also messing around with nozzle sizes and want to make sure I keep them separated, so currently my copy printers "Prusa Mini 0.6mm Nozzle" and "Prusa Mini 0.8mm Nozzle" require I re-enter the information for each printer instead of allowing it in the drop-down.

This feature might have been overlooked since most users only deal with one type of printer.

bubnikv, 2020-03-11:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3729 Please read this discussion thoroughly to understand our reasoning. 

PrusaSlicer 2.2.0-rc3 will provide you with a "Detach" button to make a copy of a system profile, which will be independent of vendor profiles, therefore it will be visible with any other profile. It is your responsibility then to ensure that the profiles are really compatible.

Closing.

closed 2020-03-11 by bubnikv

DRKMSTR, 2020-03-12:
@bubnikv Thanks for pointing me to that, thanks for the information.