prusa3d/PrusaSlicer #3821: Filament Start G-code and End G-code inserted in wrong position for MMU

michaelbeljaars, 2020-03-11:
### Version
2.1.1

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
MK3S with MMU2S

### Behavior
Start G-code and End G-code in filament settings are placed in wrong position within G-code.
Start G-code should appear **after tool change** to current filament, but  appears before. End G-code should appear **before tool change** from current filament but appears after.

[image]

I discovered this bug while playing with loading and unloading temperatures in order to get better control over the filament tips.

bubnikv, 2020-03-11:
> Start G-code should appear *after tool change* to current filament, but
appears before. End G-code should appear *before tool change* from current
filament but appears after.

That is a question. If you want to execute a block after tool change, you
may use the gool change G-code block for that and index the filament /
extruder properties using the previous_extruder / next_extruder variables.

michaelbeljaars, 2020-03-11:
> That is a question. If you want to execute a block after tool change, you may use the gool change G-code block for that and index the filament / extruder properties using the previous_extruder / next_extruder variables.

The Tool Change G-code block is executed after the unload and before the load of the next filament. It shows just directly above the Filament specific start gcode in my screenshot. What I need (and as I understand from the infobox (see other screenshot) was also intended) is to insert gcode before an unload event (lower temperature), before load (set temperature from previous filament for correct purging of the nozzle; this can be done with the Tool change gcode block) and after a load event (set correct temperature).

lukasmatena, 2020-03-11:
I think that what you're seeing is in fact the intended behaviour. Toolchange gcode was meant to be just after filament end gcode and before filament start gcode. Regardless of what the wipe tower is doing and whether there is a wipe tower at all. It is not possible to put any other custom gcode between moves generated by the wipe tower.

However, the unloading is so short that I don't see much difference between changing temperature before or after it. Unless of course I misunderstood something.

michaelbeljaars, 2020-03-13:
Yes, I agree on the placing on the Toolchange code. The bug I believe is with respect to the placing of the **Start G-code** and **End G-code** in filament settings. These are placed differently than the infobox suggests (and to where I would need them).

mbordoni, 2020-10-25:
It is even worse than expected.
By default the code sets the new nozzle temperature (the highlighted row in the image below that sets the nozzle temp for the new material) just before the ramming procedure screwing up the tip of the current filament we are going to unload.
This setting has to be postponed after the old filament has been unloaded.
Even better if we could set a specific ramming temp per filament.  

[image]

lukasmatena, 2020-10-25:
@mbordoni The gcode that you sent is fine. The temperature is changed not only after ramming, but also after retracting the filament far away from the nozzle. I see commands `G1 E-15`, then -24.5, -7 and -3.5 before the `M104`, so the tip of the filament is 50 mm far when the temperature is switched. I don't know what is screwing up your tip, but this temperature command does not.

mbordoni, 2020-10-27:
@lukasmatena thanks for your quick answer.
The point is that the new temp is set just before the cooling move and ramming are made and this badly impact the tip refining in a multi material scenario.
For example with pva you should decrease nozzle temp (from 190° to 185) before the unload to get a good tip and in our case you get even a sensible temp increase from pva to new material (getting worse and worse from pla to abs).
With this behaviour it turns out that the more cooling move you do the worse it is (since you are not cooling really the filament).
May be the possibility to set a tip temperature per filament could help in making mmu2s to work better, for sure rising temp before unloading does not help it.
Can you support us with that?

lukasmatena, 2020-10-27:
@mbordoni I'm a bit lost to be honest. First you write that "This setting has to be postponed after the old filament has been unloaded.", then explain that "For example with pva you should decrease nozzle temp (from 190° to 185) before the unload". I don't get it.

Part of the misunderstanding is that you claim that "temperature changes before ramming", which is not true. Ramming is the last extrusion just before the filament is retracted, while you clearly mean "before the part where MMU pulls the filament out completely".

Anyway, I think the discussion would just repeat https://github.com/prusa3d/PrusaSlicer/issues/1339. That issue is stale for quite some time, but I should look into it again, do some tests and see what can be done.

mbordoni, 2020-10-28:
@lukasmatena i will test it again and let you know.
Thanks by now

mbordoni, 2020-11-01:
@lukasmatena  i can confirm that the temperature change is made just after retraction  but before the cooling moves and final unload.
With different materials this means that the cooling moves are donne while the extruder temp is changing and this is not given to be a good point.
I would postpone the temp change ( using a m109 instead of the current m104) after the filament is unloaded.
This would take longer but it will be more consistent for multi material.
For specific temp for onloading and tip making i will look for and comment the correct issue.
regards

michaelbeljaars, 2020-11-01:
Speaking from my own experience, I would greatly appreciate more control over the temperature during load and unload procedures. Apart from giving more control over the shape and quality of filament tips, it also enhances the performance (and likelihood of actually functioning) when working with different types of filament (with different melting temperatures). I tend to print PVA or TPE for example quite a bit higher than PLA. In order for a proper purge when switching from PVA / TPE to PLA, I would like the temperature initially to be a bit higher than the final printing temperature for PLA.

commit 7b5158f5 (2017-03-31): Permit firmware retraction when firmware is Repetier (repetier-firmware supports it). Fixes #3821
closed 2026-09-29 by prusaslicer-bot[bot]