prusa3d/PrusaSlicer #3822: Wrong thin wall thickness

rzulian, 2020-03-11:
### Version
2.2.1+ but also 2.2.0-rc2

### Operating system type + version
macOs Catalina 10.15.2

### 3D printer brand / version + firmware version (if known)
Prusa Mk3s - 3.8.1-2869

### Behavior
* Thin wall aren't sliced accordingly to the spacing formula and are printed with the wrong thickness

I've create a project with three walls:
 1.  thickness 1.25mm wall (top left): this wall is sliced as two external perimeters and one gap fill
 2. thickness 1.27mm wall (top right): this wall is sliced as two external perimeters and one perimeter
 3. half 1.25mm and hall 1.27 wall (bottom): this wall is slice as a two external perimeters plus one long gap in the left and a small perimeter in the right side corresponding to the 1.27mm thickness
[image]

*  the thickness of a 3 perimeter wall should be 1.26

```
perimeter_spacing = extrusion_width - layer_height * (1 - π/4)
Thickness = extrusion_width + (num_perimeters-1)*perimeter_spacing
```
extrusion_width = 0.45mm
num_perimeters = 3
layer_height = 0.2mm

Perimeter_spacing  = 0.40708
Thickness = 0.45 + 0.40708*2 = 1.264

### Expected result
1.26mm thickness is not triggering a three perimeter wall, is like the 1.25mm thickness (i.e 2 external perimeters and a gap). Instead 1.27mm thickness is correct. Is this a rounding problem?

The issue is that the 1.25mm wall is printed more or less with a 1.25mm thickness, BUT the 1.27mm is printed with a 1.60mm thickness. 

[image]

rzulian, 2020-03-11:
@bubnikv   I've noticed you are working on a new version of the perimeter generator. Is this issue something that you can check?

bubnikv, 2020-03-12:
I hope the new perimeter generator will fix many issues and possible create a new set of bugs. I am just looking forward to get 2.2.0 final out of the door to have time for this perimeter fun stuff.

rzulian, 2020-03-12:
@bubnikv I'm willing to help here, but I'm struggling with the setup of the environment so far :(

swilkens, 2020-05-25:
Similar issues here, though slightly different.

Top model has 1.26 mm wall thickness, bottom model is 1.27.

Detect thin walls: OFF
Gap fill: 0 mm/s
Extrusion width: 0.45

[image]

But then, when i turn Detect thin walls ON:
[image]

closed 2026-09-29 by prusaslicer-bot[bot]