prusa3d/PrusaSlicer #3823: [2.2.1-RC] Vase mode prints base no matter what

FlailAway, 2020-03-11:
### Version
2.2.1-rc

### Operating system type + version
Mint 19.3+Cinnamon
Gen-L
Marlin 1.1.9

### 3D printer brand / version + firmware version (if known)
Anycubic delta

### Behavior
The project has an open base. There is a 6mm flat base-section at the outer rim and the vase spirals up from that. The project is basically an inverted funnel shape.

The slicer prints a full closed base. I set Bottom Layers to = 0 and still the same. It is a water-holding vase and not a funnel.

JohnnyDeer, 2020-03-12:
Hi, could you please provide 3mf with described situation? Is that happen with every model, or with one specific? Wich RC of 2.2.0 you are meaning?

bubnikv, 2020-03-12:
Actually it works, only the change in the number of bottom layers is not
processed correctly after you sliced for the first time. If you set the
number of bottom perimeters to zero, delete the object from the plater and
reload, you will get empty bottom.

bubnikv, 2020-03-12:
tested with PrusaSlicer 2.2.0-rc3.

bubnikv, 2020-03-14:
Fixed in PrusaSlicer 2.2.0-rc4.
Update after change of number of bottom layers fixed with 7f376fa927663de1cbce1df5b6aad80b202c85d0
Closing.

closed 2020-03-14 by bubnikv