prusa3d/PrusaSlicer #3827: Change options in case of 100% infill

espr14, 2020-03-12:
v2.2.0 rc3 win64

If 100% infill is set then there is a reason to change some options. Some of them don't make sense, some speed up slicing. Like fill pattern is switched to rectilinear these should be changed as well:

- extra_perimeters = false
- ensure_vertical_shell_thickness = false
- perimeters = 1
- top_solid_layers = 1
- bottom_solid_layers = 1
- infill_first = true

stefanct, 2020-10-27:
i disagree with you (e.g., having more perimeters can make sense since they are aligned) but have you even checked how 100% infill is handled in the source code? Just because it is not reflected in the GUI does not mean that the behavior isn't changed internally. Do you have a test case in mind?

espr14, 2020-10-27:
I agree that a single perimeter can be influenced by the infill so more perimeters can make the result more precise. However, there's an issue with top/bottom infill layers. You should see red and maroon parts only. The violet parts decrease strength and increase print time. I'm not sure in case of extra perimeters and ensuring vertical thickness.

Btw. printable parts (STL files) of Prusa printers might be error free, not like this part.

[infill100.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5446505/infill100.3mf.zip)

[image]

bubnikv, 2021-07-15:
Nobody else chimed in, I think we can close this unusual idea issue.

closed 2021-07-15 by bubnikv